blikvm/blikvm #3: Include copyright notice for third-party code

mtlynch, 2022-09-15:
`./src/blikvm.sh` seems to be derived from TinyPilot's copyrighted code here:

https://github.com/tiny-pilot/tinypilot/blob/609b36faf2352cb0922111b2fc9f13bd73824855/quick-install

TinyPilot's code is released [under the MIT license](https://github.com/tiny-pilot/tinypilot/blob/master/LICENSE), so you're welcome to include it, but the license requires preserving the copyright notice and license.

You can get back into compliance with the license this by copying the text from [TinyPilot's license](https://github.com/tiny-pilot/tinypilot/blob/master/LICENSE) as a header at the top of the file or by including TinyPilot's license in a file called `NOTICE` at the root of the repo.

mtlynch, 2022-09-15:
Moved to the correct repo

ThomasVon2021, 2022-09-16:
https://github.com/ThomasVon2021/blikvm/blob/master/package/kvmd-hid/disable_gadget.sh
https://github.com/ThomasVon2021/blikvm/blob/master/package/kvmd-hid/enable-gadget.sh
Thank you for reminding me. It's correct.

mtlynch, 2024-01-05:
@ThomasVon2021 - The license notice was removed in https://github.com/ThomasVon2021/blikvm/commit/74b7bd0d8df57922b05f207716411421b16dcc67#diff-13cf57767c8d201ac67a5de3c2f07904a3b64bb3e48075446c10c31f14f68c77

You're required to either remove the file or preserve the license.

ThomasVon2021, 2024-01-07:
https://github.com/ThomasVon2021/blikvm/commit/0deef067b4d728074f35037841a0f2b41acabb66
Thank you for your remind.

ThomasVon2021, 2024-01-07:
I have a question to ask you. I use GPL-3.0 license for other code, should I also add this description to all my original code?

mtlynch, 2024-08-28:
@ThomasVon2021 - I see you've again marked this as completed without actually complying with the license. Please link to where the license has been honored. Otherwise, you are still in violation of TinyPilot's MIT License.

ThomasVon2021, 2024-08-28:
[image]
[image]

ThomasVon2021, 2024-08-28:
> <img alt="截屏2024-08-29 00 08 34" width="809" src="https://private-user-images.githubusercontent.com/45224459/362367584-cf96b64b-02b6-497d-9920-971d04f506b2.png?jwt=eyJhbGciOiJIUzI1NiIsInR5cCI6IkpXVCJ9.eyJpc3MiOiJnaXRodWIuY29tIiwiYXVkIjoicmF3LmdpdGh1YnVzZXJjb250ZW50LmNvbSIsImtleSI6ImtleTUiLCJleHAiOjE3MjQ4NjE3MzYsIm5iZiI6MTcyNDg2MTQzNiwicGF0aCI6Ii80NTIyNDQ1OS8zNjIzNjc1ODQtY2Y5NmI2NGItMDJiNi00OTdkLTk5MjAtOTcxZDA0ZjUwNmIyLnBuZz9YLUFtei1BbGdvcml0aG09QVdTNC1ITUFDLVNIQTI1NiZYLUFtei1DcmVkZW50aWFsPUFLSUFWQ09EWUxTQTUzUFFLNFpBJTJGMjAyNDA4MjglMkZ1cy1lYXN0LTElMkZzMyUyRmF3czRfcmVxdWVzdCZYLUFtei1EYXRlPTIwMjQwODI4VDE2MTAzNlomWC1BbXotRXhwaXJlcz0zMDAmWC1BbXotU2lnbmF0dXJlPTI0ODNjYzA1MTkxZDBmZGNhZDdjYzg0M2QwMjhmYmIyYzdiZTNmYjczOWIzZTBmMjc5MDViZjExOTJjYjk4MDAmWC1BbXotU2lnbmVkSGVhZGVycz1ob3N0JmFjdG9yX2lkPTAma2V5X2lkPTAmcmVwb19pZD0wIn0.ssVzYNmAm9yU71DHRNogyIogG45IyIotY56rYnbGI5w"> [image]

It hasn't been modified, it's kept, right?

ThomasVon2021, 2024-08-28:
https://github.com/ThomasVon2021/blikvm/blob/master/package/kvmd-hid/enable-gadget.sh
https://github.com/ThomasVon2021/blikvm/blob/master/package/kvmd-hid/disable-gadget.sh

Still retained and not removed, please help point out the specific issues.

mtlynch, 2024-08-28:
Sorry, you're correct. I didn't realize that the link in the original comment was to an old commit rather than the latest in master.

closed 2024-08-28 by mtlynch