blikvm/blikvm #4: Fan Issue

t0mms1n, 2022-11-12:
ERROR --- Can't read fan state: Cannot connect to unix socket /run/kvmd/fan.sock

ThomasVon2021, 2022-11-12:
Because the fan model used by BLIKVM hardware is different from that of PIKVM hardware, viewing the fan status through the web interface may fail in the latest system of pikvm. However, this will not affect the normal operation of BLIKVM fan. Blikvm's software system will be launched soon, so there will be no such problems. Therefore, when using the PIKVM image, as long as you ensure that the install script displays OK, it means that the update of the kvmd fan program is successful. The threshold temperature set by the default program is 60 degrees Celsius. When the CPU temperature is detected to exceed 60 degrees Celsius, the fan will start rotating at 85% speed. If you want to modify the threshold temperature and speed, you can modify the kvmd-fan.py script, and then re-execute install.sh.

ThomasVon2021, 2022-11-12:
https://github.com/ThomasVon2021/blikvm/tree/master/package/kvmd-fan

t0mms1n, 2022-11-15:
Using the script seems to have fix the issue for me, thank you!

closed 2022-11-16 by ThomasVon2021