blikvm/blikvm #6: dev installation instructions fatal error: janus/plugins/plugin.h: No such file or directory

magore, 2022-12-09:
https://github.com/ThomasVon2021/blikvm/blob/master/dev-readme.md
ustreamer build fails looking for #include <janus/plugins/plugin.h>

Perhaps you wanted to look in  janus-gateway/src/plugins/plugin.h
There is no janus/plugins/plugin.h directory

ThomasVon2021, 2022-12-10:
Have you install janus before？

magore, 2022-12-17:
No, I have not built janus from source prior to your build dev-readme.md instructions for ustreamer.  Although, exteral to your notes,  I was aware that janus was also an available package from apt I chose the source route as you docmented it to be safe gven there were no other notes in your document.   Note: Janus itself compiled fine as you documented it.  However ustreamer  failed if we includeded your ustreamer build option  WITH_JANUS=1 - without that it builds - not sure what that breaks.

ThomasVon2021, 2022-12-26:
You should build janus before. According to instructions to config janus.

magore, 2022-12-27:
Janis was built before ustreamer following  instructions exactly ias outlined in  your dev-readme.md 
I would suggest you build everything in a clean environment and update your dev-readme.md so they work correctly.

magore, 2022-12-31:
As I said before there is no  janus/plugins/plugin.h directory just the janus-gateway/src/plugins/plugin.h. All of the other projects edit the file to correct the path. This step is missing in your dev-readme.md instructions.

freeyoung, 2023-01-08:
I encountered the same issue. Had to follow the [official docs](https://github.com/pikvm/ustreamer/blob/master/docs/h264.md#server-setup) to mitigate the compiling errors of `make WITH_JANUS=1`:

```
ln -s /opt/janus/include/janus /usr/include/janus
```

and

```
sed \
  --in-place \
  --expression 's|^#include "refcount.h"$|#include "../refcount.h"|g' \
  /usr/include/janus/plugins/plugin.h
```

ThomasVon2021, 2023-01-13:
> As I said before there is no janus/plugins/plugin.h file just the janus-gateway/src/plugins/plugin.h. All of the other KVM projects edit the file to correct the path. This step is missing in your dev-readme.md instructions.
> 
> See for example of this being fixed: https://github.com/tiny-pilot/ansible-role-ustreamer/blob/master/tasks/install_janus.yml

According to freeyoung comment, have you solved your problem?

ThomasVon2021, 2023-01-24:
https://wiki.blicube.com/blikvm/en/flashing_os/
Hi, I have share the image which can run all of software OK. You can test base on this image.

m50S79sM6SRNp8Jn, 2024-01-10:
Can this issue be closed?

closed 2024-08-28 by ThomasVon2021