blikvm/blikvm #10: OLED IP command broken

jhg03a, 2023-02-18:
There doesn't appear to be a `-I` argument to hostname on Arch.  `ifconfig | sed -En 's/127.0.0.1//;s/.*inet (addr:)?(([0-9]*\.){3}[0-9]*).*/\2/p'` will do what you want though.

MoFoQ, 2023-05-21:
yea, I noticed that too and I just edited /usr/bin/blikvm/blikvm-oled.py line 75:
`cmd = "hostname -I | cut -d\' \' -f1"`
to:
`cmd = "hostname -i | cut -d\' \' -f1"`

ThomasVon2021, 2023-05-21:
Got it.

ThomasVon2021, 2023-05-21:
> yea, I noticed that too and I just edited /usr/bin/blikvm/blikvm-oled.py line 75: `cmd = "hostname -I | cut -d\' \' -f1"` to: `cmd = "hostname -i | cut -d\' \' -f1"`

But if you are using blikvmos, you do not need to use this script to control OLED.

MoFoQ, 2023-05-22:
> But if you are using blikvmos, you do not need to use this script to control OLED.

I know but I have different issues with blikvmos (can't get features I want to work, rootfs seemed to 86 itself [ext4 took such a dump that I had to restore from image]); love the hardware tho and looking forward to the allwinner variant (as well as being able to use my older blikvm once the cm4 shortage is over)

on a related note, the OLED in blikvmos (I grabbed the "blikvm-pi-cm4-64-20230416-readonly" image) doesn't appear to show IP info; just CPU load and storage space info.

ThomasVon2021, 2023-05-22:
Are there any changes to other OLED numbers, like CPU loading? There were some issues with the April image, and the main process may crash. This issue was fixed in the May image.

closed 2024-08-28 by ThomasVon2021