blikvm/blikvm #11: OLED script permissions wrong

jhg03a, 2023-02-18:
The shell script to fork the OLED python script doesn't have execute permissions, so systemd fails.  There is a way to do exactly what the shell script does via native systemd units as an alternative.

ThomasVon2021, 2023-02-18:
Are you building a development environment? It is not clear what permissions you have to perform. The built development image has been provided.
https://wiki.blicube.com/blikvm/en/flashing_os/

jhg03a, 2023-02-18:
After a bit of juggling to get the stock image updated and the git repo baked in updated, I used the upgrade.py script which still had some issues.  The relevant thing I noticed is that the oled install script just uses a regular `cp` whereas if you look at the fan install script it uses `install` to expressly set the permissions.  So it wasn't quite a development image, but rather one trying to be upgraded in place using the release tarballs.

ThomasVon2021, 2023-02-18:
The fan script is only for PiKVM. It is not part of blikvm's own control software. I will make this clear. Thank you for reminding me. Are you using blikvm software instead of PiKVM.

jhg03a, 2023-02-20:
So the stock PCIe blikvm seems to come with a mix of PiKVM and blikvm.  Unless I'm missing something, it seems like blikvm rests on top of pikvm overriding as necessary to match the hardware in the kit.  It's possible something odd is going on just because of trying to get both up to date, but I think this is legitimate where the permissions on the oled script file in the release archive either aren't set to execute, or the install script isn't preserving that properly.  I referenced the fan script just as an alternative where it does work without extra steps.

closed 2024-08-28 by ThomasVon2021