blikvm/blikvm #15: Support for Banana Pi BPI-CM4?

melroy89, 2023-04-14:
Banana Pi has released a wonderful BPI-CM4 hardware module compatible with the Raspberry Pi cm4...

Could you verify if your image works on this device? If not. Would you please consider supporting this nice piece of cm4 hardware as well with your images?

More info: https://wiki.banana-pi.org/Banana_Pi_BPI-CM4#System_image

And on the forum: https://forum.banana-pi.org/t/banana-pi-bpi-cm4-computer-module-with-amlogic-a311d/13390

Thanks!!!

ThomasVon2021, 2023-04-14:
I am creating an H616 based image using the Armbian system. After this image, some hardware using Armbian images will be easier to port again.

melroy89, 2023-04-14:
That sounds great

melroy89, 2023-04-14:
There are btw third party images for the BPI-CM4 available, see: https://wiki.banana-pi.org/Banana_Pi_BPI-CM4#Armbian

Just wondering if that would mean your image will support this chipset as well.

So anyway, do you think I can try to do a estimated bet by buying the banana pi cm4?

Ps. I did bought your Blikvm Pcie card. So in theory hardware wise it should be compatible. Assuming you used only the first hdmi pin connector. (BPI-CM4 only support 1 hdmi instead of 2 hdmi connections with the Raspberry Pi cm4)

melroyvandenberg, 2023-04-19:
Yes or no? Should I try to order the BPI-CM4? Assuming the new image will support it?

ThomasVon2021, 2023-04-21:
The new mirror is launched with an Armbian system based on the H616 chip.

melroy89, 2023-04-21:
Uh ok.. sure, but the Banana Pi BPI Compute Module 4 is using the [Amlogic A311D chip](https://wiki.banana-pi.org/Banana_Pi_BPI-CM4#Banana_Pi_BPI-CM4_SBC_specifications).

melroy89, 2023-04-24:
Hmmm I think I will just buy a Rasp CM4.. That works at least.

ThomasVon2021, 2023-04-25:
You can buy on this link: 
https://www.aliexpress.us/item/3256805010517286.html?gps-id=pcStoreJustForYou&scm=1007.23125.137358.0&scm_id=1007.23125.137358.0&scm-url=1007.23125.137358.0&pvid=a836ab78-eeb0-4c86-8433-26e90e1be101&_t=gps-id%3ApcStoreJustForYou%2Cscm-url%3A1007.23125.137358.0%2Cpvid%3Aa836ab78-eeb0-4c86-8433-26e90e1be101%2Ctpp_buckets%3A668%232846%238113%231998&pdp_npi=3%40dis%21USD%2172.65%2172.65%21%21%21%21%21%402103200516824009886804703e0f21%2112000032095720174%21rec%21US%21251804802&spm=a2g0o.store_pc_home.smartJustForYou_6000897758043.3&gatewayAdapt=glo2usa&_randl_shipto=US

ThomasVon2021, 2023-04-25:
We have created a new hardware based on H616, called blikvm V4, which is still under testing. You can join Discord to see more information.

ThomasVon2021, 2023-04-29:
The PCIe version is designed specifically for CM4 and cannot be replaced with other modules. Because different chip pin definitions are different.

melroy89, 2023-04-29:
Maybe you should better look into the [Banana Pi BPI-CM4](https://wiki.banana-pi.org/Banana_Pi_BPI-CM4) ... It's pin compatible. It's compatible with cm4 from raspberry. 

This whole github issue is about to support that module, the pin layout isn't the problem at all. It's only a different cpu chip you might need to support in the image.

melroy89, 2023-05-03:
So in order to have HDMI input, we use CSI, so far so good. CSI is actually supported by the Banana Pi BPI-CM4 as well.. but only 1 port. And the 2nd CSI is not connected. So I hope you are not using 128-138 pins for CSI. But instead you use one of the ports in the pins 115-141?

[image]

melroy89, 2023-05-16:
Small update, I received the Banana PI CM4 module. I first tested the default Ubuntu Mate provided image from Banana Pi, which is booting from SD card. SO far so good, next will be trying to boot the BliKVM image from this hardware. Finally, I need to try to boot the image eventually from the SD card inside the BliKVM PCIe hardware.

https://www.youtube.com/watch?v=wU6zKKlJtbQ

ThomasVon2021, 2023-05-17:
I think you need build blikvm software on a armbian os or Ubuntu Mate. You can refer to https://github.com/ThomasVon2021/blikvm/blob/master/dev-readme.md. If you have any questions, we can discuss them at any time.

melroy89, 2023-05-17:
I think so as well. Raspberry Pi OS Lite is actually already Debian. But it might need the additional linux device trees files. That's it.

ThomasVon2021, 2023-05-18:
If you use Armbian, it should be relatively smooth. Currently, I have been completely successful in using Armbian on H616

melroy89, 2023-07-23:
Despite the Banana Pi CM (CM4IO) device tree dtb file being present in the `/boot/dtb/amlogic` folder of the official Banana Pi Armbian Image. These device tree files are also present in the [amlogic v6.4 / v6.5 Linux kernel](https://git.kernel.org/pub/scm/linux/kernel/git/amlogic/linux.git/).

[image]

I can not see the GPIO numbers (`gpio<number>` file) in the `/sys/class/gpio/` folder... Too bad. Which is a requirement for ArmbianIO code to get it working.

[image]

And the `gpioinfo` command gives (showing all "unnamed"):

```bash
gpiochip0 - 85 lines:
	line   0:      unnamed       unused   input  active-high 
	line   1:      unnamed       unused   input  active-high 
	line   2:      unnamed       unused   input  active-high 
	line   3:      unnamed       unused   input  active-high 
	line   4:      unnamed       unused   input  active-high 
	line   5:      unnamed       unused   input  active-high 
	line   6:      unnamed      "reset"  output   active-low [used]
	line   7:      unnamed       unused   input  active-high 
	line   8:      unnamed       unused   input  active-high 
	line   9:      unnamed       unused   input  active-high 
	line  10:      unnamed       unused   input  active-high 
	line  11:      unnamed       unused   input  active-high 
	line  12:      unnamed       unused   input  active-high 
	line  13:      unnamed       unused   input  active-high 
	line  14:      unnamed       unused   input  active-high 
	line  15:      unnamed       unused   input  active-high 
	line  16:      unnamed       unused   input  active-high 
	line  17:      unnamed       unused   input  active-high 
	line  18:      unnamed       unused   input  active-high 
	line  19:      unnamed       unused   input  active-high 
	line  20:      unnamed       unused   input  active-high 
	line  21:      unnamed       unused   input  active-high 
	line  22:      unnamed       unused   input  active-high 
	line  23:      unnamed       unused   input  active-high 
	line  24:      unnamed       unused   input  active-high 
	line  25:      unnamed       unused   input  active-high 
	line  26:      unnamed       unused   input  active-high 
	line  27:      unnamed       unused   input  active-high 
	line  28:      unnamed       unused   input  active-high 
	line  29:      unnamed       unused   input  active-high 
	line  30:      unnamed       unused   input  active-high 
	line  31:      unnamed       unused   input  active-high 
	line  32:      unnamed       unused   input  active-high 
	line  33:      unnamed       unused   input  active-high 
	line  34:      unnamed       unused   input  active-high 
	line  35:      unnamed       unused   input  active-high 
	line  36:      unnamed       unused   input  active-high 
	line  37:      unnamed      "reset"  output   active-low [used]
	line  38:      unnamed       unused   input  active-high 
	line  39:      unnamed       unused   input  active-high 
	line  40:      unnamed       unused   input  active-high 
	line  41:      unnamed       unused   input  active-high 
	line  42:      unnamed       unused   input  active-high 
	line  43:      unnamed       unused   input  active-high 
	line  44:      unnamed       unused   input  active-high 
	line  45:      unnamed       unused   input  active-high 
	line  46:      unnamed       unused   input  active-high 
	line  47:      unnamed         "cd"   input   active-low [used]
	line  48:      unnamed       unused   input  active-high 
	line  49:      unnamed       unused   input  active-high 
	line  50:      unnamed       unused   input  active-high 
	line  51:      unnamed       unused   input  active-high 
	line  52:      unnamed       unused   input  active-high 
	line  53:      unnamed       unused   input  active-high 
	line  54:      unnamed       unused   input  active-high 
	line  55:      unnamed       unused   input  active-high 
	line  56:      unnamed       unused   input  active-high 
	line  57:      unnamed      "reset"  output   active-low [used]
	line  58:      unnamed       unused   input  active-high 
	line  59:      unnamed       unused   input  active-high 
	line  60:      unnamed       unused   input  active-high 
	line  61:      unnamed       unused   input  active-high 
	line  62:      unnamed       unused   input  active-high 
	line  63:      unnamed       unused   input  active-high 
	line  64:      unnamed       unused   input  active-high 
	line  65:      unnamed       unused   input  active-high 
	line  66:      unnamed       unused   input  active-high 
	line  67:      unnamed       unused   input  active-high 
	line  68:      unnamed       unused   input  active-high 
	line  69:      unnamed       unused   input  active-high 
	line  70:      unnamed       unused   input  active-high 
	line  71:      unnamed       unused   input  active-high 
	line  72:      unnamed       unused   input  active-high 
	line  73:      unnamed       unused   input  active-high 
	line  74:      unnamed       unused   input  active-high 
	line  75:      unnamed       unused   input  active-high 
	line  76:      unnamed       unused   input  active-high 
	line  77:      unnamed       unused   input  active-high 
	line  78:      unnamed       unused   input  active-high 
	line  79:      unnamed       unused   input  active-high 
	line  80:      unnamed       unused   input  active-high 
	line  81:      unnamed       unused   input  active-high 
	line  82:      unnamed     "enable"  output  active-high [used]
	line  83:      unnamed "device-wake" output active-high [used]
	line  84:      unnamed       unused   input  active-high 
```

ThomasVon2021, 2023-09-22:
Sorry, I missed this message. I'll ask my hardware colleague for a response later.

ThomasVon2021, 2023-09-22:
You need fan、oled、atx gpio number， right？

melroy89, 2023-10-06:
I need the Banana Pi CM4 GPIO pin numbers on as a Linux device tree file. In order to use ArmbianIO.
ArmbianIO is using the old fashion device tree mapping on disk: `/sys/kernel/debug/gpio`, so without the correct device tree file, ArmbianIO software can't access the GPIO pins.

ArmbianIO is used by BlikVM (which I hope you know). Therefore, the following file also need to be extended for Banana Pi CM4 as well: https://github.com/ThomasVon2021/blikvm/blob/master/third_lib/GPIO/armbianio.c#L55 (if you know what the int values are in the array for Banana Pi then I could also [create a PR upstream project](https://github.com/bitbank2/ArmbianIO)). 

And you are correct without GPIO.. the fan, oled, atx but also simple things like a LED will _not_ work. And blikvm software will **not** start.

hzzfly, 2023-10-08:
[BliKVM-PCIe-pinout-define.xlsx](https://github.com/ThomasVon2021/blikvm/files/12839573/BliKVM-PCIe-pinout-define.xlsx)