blikvm/blikvm #20: Upgrading from 2.7.1 yields an error

marccardinal, 2023-05-17:
Current version (based on the UI):

Version: 2.7.1
Build: 2023-02-26 12:29:31

The UI reports that an update is available and fair enough, when I look at the github releases I see that indeed there is a newer version. The update however appears to be hanging or not doing anything from the UI perspective so I established an SSH connection to the unit and ran:

```
rw
sudo -i
python3 /opt/bin/blikvm/script/update.py
```

which yields the following error:

```
bash: install-kvmd-oled.sh: No such file or directory
Traceback (most recent call last):
  File "/opt/bin/blikvm/script/install_release.py", line 66, in <module>
    main()
  File "/opt/bin/blikvm/script/install_release.py", line 57, in main
    subprocess.check_output(cmd, shell = True, cwd=gArgs.releasepath )
  File "/usr/lib/python3.9/subprocess.py", line 424, in check_output
    return run(*popenargs, stdout=PIPE, timeout=timeout, check=True,
  File "/usr/lib/python3.9/subprocess.py", line 528, in run
    raise CalledProcessError(retcode, process.args,
subprocess.CalledProcessError: Command 'mount -o remount,rw / && mount -o remount,rw /boot && bash install-kvmd-main.sh && bash install-ustreamer.sh && bash install-kvmd-oled.sh && bash install-kvmd-hid.sh         && bash install-kvmd-web.sh && cp package.json /usr/bin/kvm_json/package.json &&  mount -o remount,ro / && mount -o remount,ro /boot' returned non-zero exit status 127.
Traceback (most recent call last):
  File "/opt/bin/blikvm/script/update.py", line 88, in <module>
    main()
  File "/opt/bin/blikvm/script/update.py", line 70, in main
    output = subprocess.check_output(cmd, shell = True, cwd=sh_path)
  File "/usr/lib/python3.9/subprocess.py", line 424, in check_output
    return run(*popenargs, stdout=PIPE, timeout=timeout, check=True,
  File "/usr/lib/python3.9/subprocess.py", line 528, in run
    raise CalledProcessError(retcode, process.args,
subprocess.CalledProcessError: Command 'python3 install_release.py' returned non-zero exit status 1.
```

fair enough, when I run an `ls -l` on `/tmp/kvm_update/release/` I get:

```
root@blikvm(rw):/tmp/kvm_update/release# ls -lah
total 2.4M
drwxr-xr-x 3 root root  520 May  7  2023 .
drwxr-xr-x 3 root root  120 Feb  6 01:22 ..
-rw-r--r-- 1 root root   45 May  7  2023 admin.ini
-rwxr-xr-x 1 root root  728 May  7  2023 config.ini
-rwxr-xr-x 1 root root 2.7K May  7  2023 disable-gadget.sh
-rwxr-xr-x 1 root root  528 May  7  2023 edid.txt
-rw-r--r-- 1 root root 6.9K May  7  2023 enable-gadget.sh
-rwxr-xr-x 1 root root  382 May  7  2023 install-kvmd-hid.sh
-rwxr-xr-x 1 root root  519 May  7  2023 install-kvmd-main.sh
-rwxr-xr-x 1 root root  403 May  7  2023 install-kvmd-msd.sh
-rwxr-xr-x 1 root root  768 May  7  2023 install-kvmd-web.sh
-rwxr-xr-x 1 root root  675 May  7  2023 install-ustreamer.sh
-rwxr-xr-x 1 root root  271 May  7  2023 kvmd-hid.service
-rwxr-xr-x 1 root root 227K May  7  2023 kvmd-main
-rwxr-xr-x 1 root root  304 May  7  2023 kvmd-main.service
-rw-r--r-- 1 root root 3.6K May  7  2023 kvmd-msd.sh
-rwxr-xr-x 1 root root  289 May  7  2023 kvmd-video.service
-rwxr-xr-x 1 root root  570 May  7  2023 kvmd-video.sh
-rwxr-xr-x 1 root root  285 May  7  2023 kvmd-web.service
-rwxr-xr-x 1 root root  108 May  7  2023 kvmd-web.sh
-rw-r--r-- 1 root root 1.7M May  7  2023 kvm-link
-rwxr-xr-x 1 root root  104 May  7  2023 package.json
-rw-r--r-- 1 root root  791 May  7  2023 readme.md
-rw-r--r-- 1 root root 1.7K May  7  2023 README.md
-rwxr-xr-x 1 root root 409K May  7  2023 ustreamer.bin
drwxr-xr-x 7 root root  380 May  7  2023 ventoy-1.0.88
```

Is this a bug with how the latest version is packaged or are we missing a file?

ThomasVon2021, 2023-05-17:
This should be the reason why you have already run this script with writable permissions and have fixed this issue. You can first update this script and then run it again with writable permissions.

ThomasVon2021, 2023-05-17:
This code have rm "mount -o remount,rw / && mount -o remount,rw /boot", just update /usr/bin/blikvm and try again.

marccardinal, 2023-05-17:
I ran:

```
rw
sudo -i
cd /opt/bin/blikvm
git pull
echo '{"version": "0", "version_int": 0, "md5value": "", "oled_type": 0}' > /usr/bin/blikvm/package.json
python3 /opt/bin/blikvm/script/update.py
```

this ran for a little while and finished with:

```
Upgrade success
```

I rebooted the system and it looks like that worked :)

closed 2023-05-17 by marccardinal