blikvm/blikvm #22: Popup with "Current temp CPU"

Enzokot, 2023-05-25:
As a suggestion. May be add here popup with "Current temp CPU"?
[image]

ThomasVon2021, 2023-05-25:
Ok, we will add this function. Thanks.

m50S79sM6SRNp8Jn, 2023-05-25:
Suggestion, output: 

104F / 40C

ThomasVon2021, 2023-05-25:
Ok

ThomasVon2021, 2023-05-25:
Done.
https://github.com/ThomasVon2021/blikvm/releases/tag/Develop-1.1.3

Enzokot, 2023-05-25:
@ThomasVon2021 
May be popup like this?
[image]

ThomasVon2021, 2023-05-25:
The plan is to create a graph of how many times the delay graph has been updated in the future.

ThomasVon2021, 2023-05-26:
Okay, it will be updated in the future.

m50S79sM6SRNp8Jn, 2024-01-10:
This will be addressed in our new BliKVM Matrix release, tentatively Q1/Q2 2024.

closed 2024-01-10 by m50S79sM6SRNp8Jn