blikvm/blikvm #23: IP overwrite Temp for OLED

Enzokot, 2023-05-25:
Seems after update v.1.1.3 IP isn't on the far left and again overwrite Temp 
[image]

ThomasVon2021, 2023-05-26:
Sorry I find the problem, the release file is correct, the problem is https://github.com/ThomasVon2021/blikvm/blob/master/package/kvmd-main/install-kvmd-main.sh
```
#!/bin/bash
proc_name="kvmd-main"
if [ -d "/usr/bin/blikvm" ]
then
    echo '/usr/bin/blikvm exists'
else
    mkdir -p "/usr/bin/blikvm"
fi

pid=$(ps -ef | grep "${proc_name}" | grep -v grep | awk '{print $2}')
if [ -n "${pid}" ]; then
  echo "Process $proc_name is running, killing it ..."
  kill -9 "${pid}"
fi

cp /opt/bin/blikvm/src/kvmd-main /usr/bin/blikvm/kvmd-main

mkdir -p "/usr/lib/systemd/system"
cp kvmd-main.service "/usr/lib/systemd/system"

systemctl enable kvmd-main.service
echo "install kvmd main ok"
```
this command ```cp /opt/bin/blikvm/src/kvmd-main /usr/bin/blikvm/kvmd-main``` should be ```cp kvmd-main /usr/bin/blikvm/kvmd-main```
I will fix bug right now. In the evening of Beijing time, I will send another package with version 1.1.4 to fix this bug. Could you please test it again? If everything goes well, please provide feedback. Thank you again for your feedback and help identify a bug.

ThomasVon2021, 2023-06-09:
The latest image solved this problem.

closed 2023-06-09 by ThomasVon2021