blikvm/blikvm #24: Strings for localization

Enzokot, 2023-05-26:
Can't find some strings for localization web UI (third_lib/Multi_language/en.js)
1. Text under username and button text
[image]

2. Text for popup balloon after login
[image]

3. Text for popup balloon after no update found
[image]

4. Button text on warning window for Reset ATX
[image]

5. Text for titles "Version" and "Build", button text for current version Bli KVM
[image]

6. Text for new popup with current temp and button text
[image]

ThomasVon2021, 2023-05-27:
Received, we will notify you after the repair.

ThomasVon2021, 2023-05-28:
3. Text for popup balloon after no update found
Is there any problem with this sentence? I am currently fixing it.

Enzokot, 2023-05-28:
>    Is there any problem with this sentence? I am currently fixing it.
Can't find in the file "third_lib/Multi_language/en.js" only

Enzokot, 2023-05-29:
Additionally can't find sting for empty password

[image]

ThomasVon2021, 2024-01-06:
Can you update to 1.3.9 to test. I update ui lib to solve this problem.

m50S79sM6SRNp8Jn, 2024-01-10:
1.3.9 resolved the issue

ThomasVon2021, 2024-01-28:
These issues have been resolved in the latest version. If you still find them, please resubmit the issue.

closed 2024-01-28 by ThomasVon2021