blikvm/blikvm #26: Request to Expand Partition on both eMMC and SD Card

Alsa2, 2023-07-09:
Hello,

I recently acquired a [cm4 pcie blicube blikvm](https://www.blicube.com/blikvm-pcie/) and successfully installed blikvm on the eMMC of the cm4. Additionally, I inserted an SD card into the device to increase the available storage capacity.

Following the instructions provided in the [expanding the file system](https://wiki.blicube.com/blikvm/en/Expand_Filesystem/#method-3-use-script) guide, I ran the script to expand the file system. However, I noticed that the expansion only occurred on the eMMC and not on the inserted SD card.

I would like to request assistance in expanding the partition on both the eMMC and the available SD card. Could you please guide me on how to achieve this? Thank you!

ThomasVon2021, 2023-07-09:
I don't know about this at the moment. I will continue my studies and if there is any progress, I will reply to you here

ThomasVon2021, 2023-07-15:
https://discord.com/channels/943534043515977768/943534839213228125/1129799253632491620
By reviewing the official documentation of CM4, it can be seen that several pins of the SD card are only valid in the lite version of the CM4 hardware design.

Alsa2, 2023-07-15:
I see, could be useful to add this in the documentation so people buying the cm4's are aware of this.
Thank you very much for your help!

ThomasVon2021, 2023-07-16:
Ok, I will add this to faq.

closed 2023-07-16 by Alsa2