blikvm/blikvm #34: [Feature Request] Settings new features

NeVa0Theresia, 2023-07-24:
+Reboot option for BliKVM itself
+SSH shell support

ThomasVon2021, 2023-07-31:
add Reboot option;

ThomasVon2021, 2024-01-07:
We will start considering adding SSH shell this year

NeVa0Theresia, 2024-01-07:
> We will start considering adding SSH shell this year

Really appreciate it. Thanks

m50S79sM6SRNp8Jn, 2024-09-08:
Good news, these two features will be made available fairly soon.

m50S79sM6SRNp8Jn, 2024-09-20:
Recommend you upgrade to latest version v.1.4.6-alpha (or higher). The features are available there.

NeVa0Theresia, 2024-09-20:
Thanks to the dev team! Featrures have been added to the new ui. I updated v4 from 1.4.0 to 1.4.x, the redesigned web ui would clear the previous web ui config, especially web login credentials; therefore i recommand displaying a warning or notification during the update on the terminal.

ThomasVon2021, 2024-09-21:
> 感谢开发团队！新 UI 中添加了功能。我将 v4 从 1.4.0 更新到 1.4.x，重新设计的 Web UI 将清除以前的 Web UI 配置，尤其是 Web 登录凭据；因此我建议在终端上更新期间显示警告或通知。

Will add warn, thank you.

closed 2024-09-21 by ThomasVon2021