blikvm/blikvm #43: blikvm crash with intel nuc 12 pro

to4ko, 2023-08-20:
Trying to run blikvm (edid for 1080p50hz) with my nuc 12 pro  leads to blikvm restart while nuc shutsdown. More over, nuc cannot be started after that with power button. Only psu disconnection helps. Nothing in logs. Sametime blikvm works perfectly fine with other my pc.

ThomasVon2021, 2023-08-21:
BlikVM software have some progress, like kvmd-main and kvm-link, can you tell which program has a problem when you manually execute it?

to4ko, 2023-08-22:
> BlikVM software have some progress, like kvmd-main and kvm-link, can you tell which program has a problem when you manually execute it?

Will do ASAP - right now this NUC in production and there is no options to "test". Soon new NUC will arrive and i will do tests.

ThomasVon2021, 2023-08-23:
Okay, waiting for your results.

to4ko, 2023-09-01:
so. if USB is not connected - no problem at all. once USB connected to NUC (USB 2.0 port) - everything shuts down

if i run kvmd-main

[image]

and everything shuts down

to4ko, 2023-09-01:
same behaviour with USB2.0 port and USB3.0 port.

any specific settings might be applied in NUC BIOS?

ThomasVon2021, 2023-09-05:
Can you use gdb run, to get the bt?

to4ko, 2023-09-05:
> Can you use gdb run, to get the bt?

could you please elaborate a little bit - i don't get what i need to do

to4ko, 2023-09-07:
as i proposed - it's all about usb settings in nuc. all kind of charging supposed to be turned off for port used to connect with blikvm

closed 2023-09-07 by to4ko