blikvm/blikvm #52: Where ist the override.yaml in BliKVM v4 Allwinner

zag1productions, 2023-09-07:
We have recently ordered 3 BliKVMv4 Allwinner and need to change the USB-Mode to disable mass storage in order to work with Dell Desktop BIOS. Where in the filesystem should we place or can we find the override.yaml to add the following settings:

kvmd:
	hid:
		mouse_alt:
			device: /dev/kvmd-hid-mouse-alt
	msd:
		type:  disabled

Thanks a lot! 

Florian

ThomasVon2021, 2023-09-07:
If you not use kvmd-msd.sh, mass storage will not disable. If you want to disable or clean mass storage, you can check the guide:
https://github.com/ThomasVon2021/blikvm/tree/master/package/kvmd-msd

zag1productions, 2023-09-07:
Hi Thomas, thank you! 
This was not specific what i asked for, but we will see if it works. There is a generic usb-mode, that supports under any circumstances only mouse/keyboard (HID) and no storage device within the same port/cable. this mode is mostly compatible and has (so my understanding) nothing to do with the storage mounting or unmounting you described in your answer. 

Have a great day.

closed 2023-09-07 by zag1productions