blikvm/blikvm #54: HDMI swithc issue

to4ko, 2023-09-13:
i'm running blikvm hat with hdmi kvm switch. but no control from blikvm gui available. also there is no ttyUSB0 device appearing once control usb cable connected. is there any tricks like special usb cable or smthg else?

[image]

ThomasVon2021, 2023-09-14:
https://wiki.blicube.com/blikvm/en/switch-guide/#introduction
Yes, there are special wires, and there is a wire labeled kvm for this purpose. Please check the specific wiring on the wiki, and if convenient, you can take photos.

to4ko, 2023-09-14:
Ah. I see. Most probably it's all about the "special" canle. Any chance to buy it or get pinouts/schemes to make my own?

ThomasVon2021, 2023-09-14:
Our packaging is already equipped with this kvm control line and labeled, isn't it in your package?

to4ko, 2023-09-14:
unfortunately i don't have it as i used switch from another project. it was already on hand and there was no reason to buy new one.

ThomasVon2021, 2023-09-14:
Does your switch support remote switching? If the switch you are using does not support control itself, then there is no way.
If it is supported by itself, but the protocol is different, it can be supported through software development.

to4ko, 2023-09-14:
i't same XH-HK4401 as one you selling. only thing i'm missing is "USB remote cable"

[image]

ThomasVon2021, 2023-09-14:
Currently, there are no products purchased separately from this line, as cross-border express delivery is expensive and it is difficult to sell on a separate line.

to4ko, 2023-09-15:
ok. i et you point. will make my own cable.

closed 2023-09-15 by to4ko

to4ko, 2023-09-24:
solved with locally purchased ftdi module