blikvm/blikvm #55: Wake-On-Lan feature

to4ko, 2023-09-18:
Hi!

as I can see there is wake-on-lan feature mentioned. So is it possible to wake connected pc and how? or it's about wakig kvm itself?

ThomasVon2021, 2023-09-19:
Adding 'walk on lan' is technically feasible, but I didn't know much about this technology before. I would like to ask you why you need this feature?

n3gwg, 2023-09-19:
Are you serious?

The BliKVM can be powered by an USB battery or UPS for a long time. Therefore, being 5v and low current it could be the only thing left running when the power comes back on. Some servers need to be told to power back on. Hence it can succeed doing this through wake on lan to other servers very easily.

The same can be said of the BliKVM itself. Wake on lan packets can be sent via a Cron job so any server that stays alive, send wake on lan packets to every other server that did not survive the power loss and did not reboot when the power came back up.

Stuart

ThomasVon2021, 2023-09-19:
Okay, thank you for your guidance. The function of the 'walk on lan' test version has been developed and will be updated in subsequent versions after passing the test.

n3gwg, 2023-09-19:
Developers, et alia:

Desirous of instantiating my greatest appreciation, let me state that is most excellent news!

Stuart

ThomasVon2021, 2023-09-22:
https://wiki.blicube.com/blikvm/en/wake-on-lan/#introduction
Update to 1.2.1.
I test ok in my end, you can test it.

closed 2023-09-24 by ThomasVon2021