blikvm/blikvm #56: Multiple SSID BLIKVM PCIe

rizwaniqbal76, 2023-09-24:
connect to WIFI  A and ping defined monitoring IP example 8.8.8.8, BLIKVM should keep the WIFI connection connected until defined quality parameters, example if we get 50% Packet loss, BLIKVM should disconnect SSID A and will connect to SSID B, and again will check quality of internet, Example if defined quality threshold is 50% ,  If 50% packets are lost from total sent packets BLIKVM should connect to next SSID C .  We should be able to  define 2 SSID or 10 SSID in the config file
The purpose of the above requirement is to keep the BLIKVM always online, If any further questions please send a message

ThomasVon2021, 2023-09-24:
How do you judge 50% Packet loss？ I can ping defined ip or use some command to check wifi strength. A config json like：
[
    {
        "SSID": "YourSSID1",
        "Password": "YourPassword1",
        "IP": "192.168.1.1"
    },
    {
        "SSID": "YourSSID2",
        "Password": "YourPassword2",
        "IP": "192.168.1.2"
    },
    {
        "SSID": "YourSSID3",
        "Password": "YourPassword3",
        "IP": "192.168.1.3"
    }
]

rizwaniqbal76, 2023-09-25:
We don't need to check signal strength , Most important is that internet is stable. We can calculate quality of internet by ping reply . If there is packet loss , we have to move to next WIFI   , i attached some images from OPENWRT , just to have an idea , 

If we will connect only based on signal strength , and internet is not available on connected  SSID , IPKVM will not work , Even if signals are 100 percent if internet is having packet loss , we have to move to next SSID .

[image]
[image]

ThomasVon2021, 2023-09-25:
Ok, got it like this.
terminal ~ % ping -c 10 192.168.0.107
PING 192.168.0.107 (192.168.0.107): 56 data bytes
64 bytes from 192.168.0.107: icmp_seq=0 ttl=64 time=79.653 ms
64 bytes from 192.168.0.107: icmp_seq=1 ttl=64 time=7.210 ms
64 bytes from 192.168.0.107: icmp_seq=2 ttl=64 time=339.452 ms
64 bytes from 192.168.0.107: icmp_seq=3 ttl=64 time=6.445 ms
64 bytes from 192.168.0.107: icmp_seq=4 ttl=64 time=279.133 ms
64 bytes from 192.168.0.107: icmp_seq=5 ttl=64 time=20.583 ms
64 bytes from 192.168.0.107: icmp_seq=6 ttl=64 time=7.014 ms
64 bytes from 192.168.0.107: icmp_seq=7 ttl=64 time=7.572 ms
64 bytes from 192.168.0.107: icmp_seq=8 ttl=64 time=50.521 ms
64 bytes from 192.168.0.107: icmp_seq=9 ttl=64 time=2.507 ms

--- 192.168.0.107 ping statistics ---
10 packets transmitted, 10 packets received, 0.0% packet loss
round-trip min/avg/max/stddev = 2.507/80.009/339.452/117.752 ms

Do you think the above configuration file is suitable? I think I will write a Python file, and once the basic logic is written, it will be easier for you to maintain it yourself later.

rizwaniqbal76, 2023-09-26:
Yes , it will work , we have to check connected WIFI internet Quality ,   ping is the only way for us to see if SSID is connected to internet ,  I need complete configuration file which can work according to requirements  , In above file you are just sending ping , but we also have to make decision about connection , it should be connected or move to next SSID

ThomasVon2021, 2023-09-27:
config file means:
[
{
"SSID": "YourSSID1",
"Password": "YourPassword1",
"IP": "192.168.1.1"
},
{
"SSID": "YourSSID2",
"Password": "YourPassword2",
"IP": "192.168.1.2"
},
{
"SSID": "YourSSID3",
"Password": "YourPassword3",
"IP": "192.168.1.3"
}
]
it is a json file. The specific strategy code has been basically clarified and will be written.

ThomasVon2021, 2023-10-25:
https://github.com/ThomasVon2021/blikvm/tree/master/package/multiple-SSID
You can test .

closed 2023-10-27 by ThomasVon2021