blikvm/blikvm #60: show Pi status information in pikvm web interface

Chavell3, 2023-10-05:
a great additional feature would be to show the status information that is shown on the front OLED and also the fan status directly on the pikvm web interface directly. 
So in that case you have the health status of the system also visible while looking at other systems.

BR
Thomas

Chavell3, 2023-10-05:
I saw at least the cpu temperature can be found under, System -> About -> Hardware already. but it would be grate to have that information available togehter with the disk utilization and fan speed(or whatever fan information is readable) directly on the left side of the top bar within e.g. pikvm

m50S79sM6SRNp8Jn, 2023-10-06:
Ok yes I see how this will be useful. We also have some additional thoughts in that direction.

blikvm, 2025-08-04:
[image]
In version v2.0.4, the display of this information has been added. If you need more information, you can reopen the issue.