blikvm/blikvm #61: [Feature Request] Add 'Mouse polling rate' option

kywarai, 2023-10-06:
Feature Request: Add a Mouse Polling Rate option, similar to that in PiKVM, for adjusting the polling rate of mouse actions. 
Shortening the polling interval can make mouse movement smoother.

Here's the option in the PiKVM, the option in PiKVM is actually an interval time, the smaller the value, the higher the polling rate.
[image]

ref: https://github.com/pikvm/kvmd/blob/988a190957a367ed7a4b177b428507135d37709c/web/share/js/kvm/mouse.js#L115

m50S79sM6SRNp8Jn, 2024-01-10:
We plan to deliver this feature as part of our new BliKVM Matrix release, tentatively Q1/Q2 2024.

ThomasVon2021, 2024-02-19:
Considering network transmission, the mouse speed is currently adaptive, so this feature will not be added temporarily.

closed 2024-02-19 by ThomasVon2021