blikvm/blikvm #63: Feature request: add a "hard power off" or "press and hold power" option

harlong, 2023-10-06:
[image]
Can you add an option to "Press and hold power switch" or "hard power off"? In most ATX systems to turn off the stalled PC i must hold "Power" for at least 3-5 seconds

ThomasVon2021, 2023-10-06:
Doe now can't power off? The latest version atx poweroff default time is 5 seconds.

ThomasVon2021, 2023-10-06:
Ok, If you update to the latest version, still work, I think maybe I put the time config will be also ok?

harlong, 2023-10-06:
Yes, it may be a good idea

harlong, 2023-10-06:
Yes, i tested on dev-1.1.8, then it was 2 seconds

ThomasVon2021, 2023-10-07:
when you update to 1.2.0，is it ok？

harlong, 2023-10-07:
I think it is ok, but i can not check, because my server is not freezing anymore

ThomasVon2021, 2023-11-19:
You can change the power on or off delay time in the 1.2.7 version. The config file is in /usr/bin/blikvm/package.json

closed 2023-11-19 by ThomasVon2021