blikvm/blikvm #64: Web UI issues (Develop-1.2.2)

harlong, 2023-10-06:
[image]
1. Button and warning messages on login form are not translated

[image]
2. Buttons on reset confirmation form are not translated

[image]
3. Button on version information form is not translated

4. In the /opt/bin/blikvm/third_lib/Multi_language there is a French translation, but it's unavailable in language settings.
5. I have sent you a russian translation, but it's too yes not included

harlong, 2023-10-06:
[ru.js.zip](https://github.com/ThomasVon2021/blikvm/files/12833413/ru.js.zip)

ThomasVon2021, 2023-10-07:
I will check it, thank you.

melroy89, 2023-11-03:
I have the same issue. Also those labels as still Chinese. Even the notifications I get are still in Chinese:

[image]

melroy89, 2023-11-09:
https://github.com/ThomasVon2021/blikvm/issues/24 is related.

Do you know the root-cause already?

devzwf, 2023-11-19:
same in French
would be interesting to update those

harlong, 2023-11-20:
Upgraded to dev-1.2.8:
- When checking for updates, popup "No higher version fetched" is not translated
- "Power ON" label in web GUI should be changed to "On/Off" (see discord "dev" chat)
- "Power OFF" label should be changed to "Force off" (see discord "dev" chat)
- After "Version information:" in "More" screen space is missing
- In "Send text" menu, caption is not translated (see pic belos)
- "In development" in MSD is not translated
- In change password form after "strength:' space is missing
- On login form two warning messages and confirmation button are not translated (see pic below)
- "Successful login" notice is not translated (see pic below)
- Text in "Device is offline" warning is not wrapped to the next line, and is just cut off (see pic below)
- "Language" in additional menu is not translated
- Russian text width is still somehow weird, i don't know, how to shorten it (see pic below)
- Anyhow, i've committed an updated ru.js

[image]
[image]
[image]
[image]
[image]
[image]

harlong, 2023-11-20:
One more Chinese text that is not translated

[image]

ThomasVon2021, 2024-08-25:
https://wiki.blicube.com/blikvm/en/flashing_os/
You can flash the lasted image to test.

closed 2024-08-25 by ThomasVon2021

melroy89, 2024-08-25:
Or by upgrading, right? https://wiki.blicube.com/blikvm/en/update/

ThomasVon2021, 2024-08-25:
yes, use alpha