blikvm/blikvm #65: FAN is getting different speed setting every second

Chavell3, 2023-10-16:
Hi Team,

I just recently found from the log, that the FAN speed is set multiple times within just a second.
Maybe that could be changed, so maybe just every 5min or 10min the temperature is checked and the FAN speed is adjusted.
I wouldn't think that this would be a risk for the CPU. But it would probably increase the FAN's lifetime.

Or I think maybe the temperature is checked every minute and if the CPU temp raises like 6 or 7 Celsius(as there is always some small mismatch of like 1Celsius when reading it 2 times one after the other) then adjust the FAN speed

Example from my log:
```
[2023-10-16 10:37:05 kvmd-fan.service] --- Temp: 46.251
[2023-10-16 10:37:05 kvmd-fan.service] --- set_speed: 31 %
[2023-10-16 10:37:05 kvmd-fan.service] --- Temp: 45.764
[2023-10-16 10:37:05 kvmd-fan.service] --- set_speed: 28 %
[2023-10-16 10:37:05 kvmd-fan.service] --- Temp: 45.764
[2023-10-16 10:37:05 kvmd-fan.service] --- set_speed: 28 %
[2023-10-16 10:37:05 kvmd-fan.service] --- Temp: 46.251
[2023-10-16 10:37:05 kvmd-fan.service] --- set_speed: 31 %
[2023-10-16 10:37:05 kvmd-fan.service] --- Temp: 48.199
[2023-10-16 10:37:05 kvmd-fan.service] --- set_speed: 40 %
[2023-10-16 10:37:05 kvmd-fan.service] --- Temp: 46.738
[2023-10-16 10:37:05 kvmd-fan.service] --- set_speed: 33 %
[2023-10-16 10:37:05 kvmd-fan.service] --- Temp: 46.251
[2023-10-16 10:37:05 kvmd-fan.service] --- set_speed: 31 %
[2023-10-16 10:37:05 kvmd-fan.service] --- Temp: 46.251
[2023-10-16 10:37:05 kvmd-fan.service] --- set_speed: 31 %
[2023-10-16 10:37:05 kvmd-fan.service] --- Temp: 47.712
[2023-10-16 10:37:05 kvmd-fan.service] --- set_speed: 38 %
[2023-10-16 10:37:05 kvmd-fan.service] --- Temp: 45.764
[2023-10-16 10:37:05 kvmd-fan.service] --- set_speed: 28 %
[2023-10-16 10:37:05 kvmd-fan.service] --- Temp: 46.251
[2023-10-16 10:37:05 kvmd-fan.service] --- set_speed: 31 %
[2023-10-16 10:37:05 kvmd-fan.service] --- Temp: 45.764
[2023-10-16 10:37:05 kvmd-fan.service] --- set_speed: 28 %
[2023-10-16 10:37:05 kvmd-fan.service] --- Temp: 46.738
[2023-10-16 10:37:05 kvmd-fan.service] --- set_speed: 33 %
[2023-10-16 10:37:05 kvmd-fan.service] --- Temp: 46.251
[2023-10-16 10:37:05 kvmd-fan.service] --- set_speed: 31 %
[2023-10-16 10:37:05 kvmd-fan.service] --- Temp: 45.277
[2023-10-16 10:37:05 kvmd-fan.service] --- set_speed: 26 %
[2023-10-16 10:37:05 kvmd-fan.service] --- Temp: 46.251
[2023-10-16 10:37:05 kvmd-fan.service] --- set_speed: 31 %
```

Thanks.

ThomasVon2021, 2023-10-16:
Which image do you use（pikvm or blikvm）？ Which hardware do you have？(v1 v2 v3 or v4)

Chavell3, 2023-10-17:
I use the Pikvm image and the v4 Hat hardware

ThomasVon2021, 2023-10-18:
https://github.com/ThomasVon2021/blikvm/blob/master/package/kvmd-fan/kvmd-fan.py
You can update to use this code.

Chavell3, 2023-10-25:
Now I got that error after changing the config.
[2023-10-25 17:03:11 kvmd-fan.service] --- Temp:
```
[2023-10-25 17:03:12 kvmd.service] --- aiohttp.server                   ERROR --- Error handling request
[2023-10-25 17:03:12 kvmd.service] --- Traceback (most recent call last):
[2023-10-25 17:03:12 kvmd.service] ---   File "/usr/lib/python3.11/site-packages/aiohttp/web_protocol.py", line 433, in _handle_request
[2023-10-25 17:03:12 kvmd.service] ---     resp = await request_handler(request)
[2023-10-25 17:03:12 kvmd.service] ---            ^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^
[2023-10-25 17:03:12 kvmd.service] ---   File "/usr/lib/python3.11/site-packages/aiohttp/web_app.py", line 504, in _handle
[2023-10-25 17:03:12 kvmd.service] ---     resp = await handler(request)
[2023-10-25 17:03:12 kvmd.service] ---            ^^^^^^^^^^^^^^^^^^^^^^
[2023-10-25 17:03:12 kvmd.service] ---   File "/usr/lib/python3.11/site-packages/aiohttp/web_middlewares.py", line 117, in impl
[2023-10-25 17:03:12 kvmd.service] ---     return await handler(request)
[2023-10-25 17:03:12 kvmd.service] ---            ^^^^^^^^^^^^^^^^^^^^^^
[2023-10-25 17:03:12 kvmd.service] ---   File "/usr/lib/python3.11/site-packages/aiohttp/web_middlewares.py", line 108, in impl
[2023-10-25 17:03:12 kvmd.service] ---     return await handler(request)
[2023-10-25 17:03:12 kvmd.service] ---            ^^^^^^^^^^^^^^^^^^^^^^
[2023-10-25 17:03:12 kvmd.service] ---   File "/usr/lib/python3.11/site-packages/kvmd/htserver.py", line 324, in wrapper
[2023-10-25 17:03:12 kvmd.service] ---     return (await exposed.handler(request))
[2023-10-25 17:03:12 kvmd.service] ---             ^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^
[2023-10-25 17:03:12 kvmd.service] ---   File "/usr/lib/python3.11/site-packages/kvmd/apps/kvmd/api/log.py", line 57, in __log_handler
[2023-10-25 17:03:12 kvmd.service] ---     await response.write(("[%s %s] --- %s" % (
[2023-10-25 17:03:12 kvmd.service] ---   File "/usr/lib/python3.11/site-packages/aiohttp/web_response.py", line 514, in write
[2023-10-25 17:03:12 kvmd.service] ---     await self._payload_writer.write(data)
[2023-10-25 17:03:12 kvmd.service] ---   File "/usr/lib/python3.11/site-packages/aiohttp/http_writer.py", line 115, in write
[2023-10-25 17:03:12 kvmd.service] ---     self._write(chunk)
[2023-10-25 17:03:12 kvmd.service] ---   File "/usr/lib/python3.11/site-packages/aiohttp/http_writer.py", line 75, in _write
[2023-10-25 17:03:12 kvmd.service] ---     raise ConnectionResetError("Cannot write to closing transport")
[2023-10-25 17:03:12 kvmd.service] --- ConnectionResetError: Cannot write to closing transport
```

ThomasVon2021, 2023-10-25:
You can take a look at the python code, it's very simple. If you want to add more monitoring, I think it should be simple too. 
It's not quite sure what the previous version of the fan program you have is, but there is no problem in updating to the latest version of the current github repository.

Chavell3, 2023-10-30:
I had a look today into my log files, the issue persists although I used your latest config. 
It seems that is only appears the time that script runs, that it checks and adjusts the fan speed multiple times in a second...
weird

ThomasVon2021, 2023-10-31:
In the code, you can see 
```
time.sleep(5)
```
That mean every loop will sleep 5s, I don't know how this time do you get:
```
[2023-10-16 10:37:05 kvmd-fan.service] 
```
The script don't have print time.

closed 2025-04-17 by ThomasVon2021