blikvm/blikvm #67: 增加关闭oled开关，或者调整显示间隔防止烧屏

vileer, 2023-11-02:
oled使用久了会烧屏，最好增加oled关闭开关，同时为了避免用户忘记ip，关闭逻辑最好改为每次机器上电屏幕显示一定时常，显示完才判断开关状态，如果为关闭则不显示，为开则继续显示。
这样要是用户忘记IP信息至少通过简单重新上电再次获取信息，而无需重刷系统或者登陆路由器查看。

klzsysy, 2023-11-02:
补充：
可以再增加一个模式：在正常运行过程中，可以3~5分钟显示一次，一次显示一个循环，一是避免需要重启才能看到信息，二是完全不显示长时间运行如死机无法发现

m50S79sM6SRNp8Jn, 2023-11-02:
This what I have understood:

"To ensure users don't forget the IP address, it's advisable to modify the shutdown logic. Whenever the machine is powered on and the screen is displayed intermittently, the system should check the switch status after displaying. If the switch is off, no IP information is shown; if it's on, the IP information continues to display. This approach allows users to retrieve IP information by simply restarting the machine without the need to reflash the system or log in to the router for verification. Additionally, you can introduce another mode: during regular operation, display the IP information every 3-5 minutes in a single cycle. This serves the dual purpose of preventing the need for a full system restart to view the information and ensuring that the information isn't displayed continuously, especially in the event of a system crash that remains unnoticed."

What is the problem you are facing today? 
Can I please ask you to elaborate so I can fully understand?

vileer, 2023-11-03:
@m50S79sM6SRNp8Jn The key point of this issue is to have a way to turn the screen off. Because some of the users have burned their OLED screens after long-term use, see the following picture, the area of the time and temperature had been burned. The pieces of advice @klzsysy  and I suggested are making sure the users have a way to get the IP information after turning off their screens for a long time. You don't have to follow the suggestions if you have more better elegant way to let the users get the IP information.
[image]

m50S79sM6SRNp8Jn, 2023-11-03:
It's clear now, thanks for your clarifications. Let me think about a suitable solution.

srepac, 2023-11-07:
You can turn off and on the oled as needed.  Here are two scripts I wrote to do just that.
```
[root@epikvm KVM-Backups]# cat oled-off
#!/bin/bash
###
# This script was created to completely turn OFF the PiKVM oled
###

# First, disable and stop kvmd-oled service
systemctl disable --now kvmd-oled

# Then, show oled status message for 1 sec and clear the oled on exit
/usr/bin/kvmd-oled --height=32 --interval=1 --clear-on-exit --text="Turning OFF oled"
printf "\nThe OLED should be completely OFF now.\n"

[root@epikvm KVM-Backups]# cat oled-on
#!/bin/bash
###
# This script was created to turn ON the PiKVM oled
###
# Enable kvmd-oled-reboot and -shutdown services
systemctl enable kvmd-oled-reboot kvmd-oled-shutdown

# Enable AND Start kvmd-oled service
systemctl enable --now kvmd-oled
if [ $? -eq 0 ]; then printf "\nThe OLED should be ON now.\n"; fi
```

I hope this helps.

ThomasVon2021, 2023-11-08:
> You can turn off and on the oled as needed. Here are two scripts I wrote to do just that.
>
> ```
> [root@epikvm KVM-Backups]# cat oled-off
> #!/bin/bash
> ###
> # This script was created to completely turn [...]

So pikvm both write kvmd-oled-reboot and kvmd-oled-shutdown services，not only kvmd-oled services.

srepac, 2023-11-08:
> So pikvm both write kvmd-oled-reboot and kvmd-oled-shutdown services，not only kvmd-oled services.

kvmd-oled-reboot and kvmd-oled-shutdown came about as a result of my contribution to show something on the oled when rebooting or shutting down the pikvm.

ThomasVon2021, 2023-11-08:
> kvmd-oled-reboot and kvmd-oled-shutdown came about as a result of my contribution to show something on the oled when rebooting or shutting down the pikvm.

So if I want to realize these function in the blikvm software, I need also to write some code support reboot and shutdown.

srepac, 2023-11-08:
> > kvmd-oled-reboot and kvmd-oled-shutdown came about as a result of my contribution to show something on the oled when rebooting or shutting down the pikvm.
> 
> So if I want to realize these function in the blikvm software, I need also to write some code support reboot and shutdown.

Correct

ThomasVon2021, 2023-11-19:
If you use blikvm os, you can update to 1.2.7 to test.

closed 2023-11-19 by ThomasVon2021