blikvm/blikvm #71: Move the keyboard a bit up (y-axis)

melroy89, 2023-11-07:
With the new keyboard layout (with marcos) being a bit bigger (in height), the keyboard is not showing up fully on my screen by default:

[image]

Move the default position a bit higher (y-axis).. So it should look something like this by default instead:

[image]

melroy89, 2023-11-07:
Also the background blur effect does not fit anymore:

[image]

Where can I find this keyboard.vue project file, containing the new short cuts/ marcos? This I can't find in this repository but I also can't find it in https://github.com/ThomasVon2021/blikvm-webclient repository.

ThomasVon2021, 2023-11-08:
Thank you for your feedback, I will fix this issue as soon as possible. The old code is in the blikvm-webclient repository. But the latest code is not available. We are currently refactoring the code, and after refactoring, we plan to push to the github.

ThomasVon2021, 2023-11-08:
> Also the background blur effect does not fit anymore (as you can see the buttons at the bottom are leaving the background):

I don’t quite understand this question. Can you explain it in more detail?

melroy89, 2023-11-08:
Sorry if I wasn't clear. The keyboard has this background effect (this white blurry background), right? But by introducing the extra macro keys at the top of the keyboard, the size of the keyboard increase. However, you forgot to increase the background blur effect to increase as well. Making the keyboard fit in the background again.

[image]

ThomasVon2021, 2023-11-08:
> Sorry if I wasn't clear. The keyboard has this background effect (this white blurry background), right? But by introducing the extra macro keys at the top of the keyboard, the size of the keyboard increased. However, you forgot to increase the background blur effect to increase as well. Making the keyboard fit in the background again.
> 
> Now the last row doesn't fit anymore in the background. It's just minor CSS design detail.
> 
> [image]

Clear now, thanks.

devzwf, 2023-11-14:
Also, 
would be awesome , to have different Keyboard depending on language ? or not ?
en keyboard ,  fr keyboard , etc.....  (qwerty / azerty / etc...  )

ThomasVon2021, 2023-11-15:
You can update to 1.2.6.

melroy89, 2023-11-16:
> Also, would be awesome , to have different Keyboard depending on language ? or not ? en keyboard , fr keyboard , etc..... (qwerty / azerty / etc... )

Agreed, I think for now only qwerty English language is supported.