blikvm/blikvm #79: [Feature Request] Support Relative Mouse Mode

zegelin, 2023-11-16:
When trying to use a BliKVM v4 connected to a Hikvision NVR I can see the video output of the NVR and while the mouse/keyboard works, the local client mouse position is totally off compared to the remote pointer.  It's only correct when the local mouse is positioned to the very top-right corner of the screen.

I note that PiKVM docs talk about mouse modes (https://docs.pikvm.org/mouse/), but I cannot see any mention of that in the BliKVM v4 docs.  I've ssh'ed into the KVM but `/etc/kvmd` doesn't exist.

Tried in Firefox and Chrome. Same problem. Also tried different aspect ratio settings. 

First screenshot shows the local and remote pointers closely aligned in the top-right.  Second screenshot shows local cursor near the bottom-left, yet remote cursor is in the middle of the screen.

[image]

[image]

I asked on the BliKVM Discord if this feature was available and it was explained to me that v4 currently doesn't support this feature.

Can relative mouse mode be added to the v4 device OS?

m50S79sM6SRNp8Jn, 2024-01-10:
We plan to deliver this feature as part of our new BliKVM Matrix release, tentatively Q1/Q2 2024.

dahappychappy, 2024-05-27:
I am having the exact same issue.
[image]

ThomasVon2021, 2024-05-28:
> I am having the exact same issue.

Are you clear about the difference between relative mouse and absolute mouse modes? I am currently developing a new architecture code. If you are familiar with it, I would like to open it up to you first and add this feature together.

benalexau, 2024-05-28:
I am also having the same issue, also with a Hikvision NVR. Happy to test code.

ThomasVon2021, 2024-08-25:
https://wiki.blicube.com/blikvm/en/flashing_os/
You can flash the lasted image to test.

benalexau, 2024-09-06:
Tested with  v1.4.3 alpha but had mixed results.

The mouse pointer movement in relative mode seems to be OK in Firefox. As you move the mouse pointer around it appears to match what is shown in the browser versus that implied by the target system enabling / disabling "mouse over" controls. However, clicking the mouse button does not register as a click on the target system. I found you can work around that somewhat by toggling back to absolute then relative mode (skipping the reboot prompt), at which point you can click once only. I encountered similar behaviour in Chrome, although I didn't spend time looking for workarounds given I mainly use Firefox.

ThomasVon2021, 2024-09-06:
Did the KVM and controlled PC restart after switching modes?

ThomasVon2021, 2024-09-06:
> Yes, rebooted both after setting to relative mode. Also rebooted again just then and same situation.
> 
> I did find in Chrome if I repeatedly click then sometimes it works.

ok, I have reproduced this issue and will inform you after fixing it

ThomasVon2021, 2024-09-09:
> Yes, rebooted both after setting to relative mode. Also rebooted again just then and same situation.
> 
> I did find in Chrome if I repeatedly click then sometimes it works.

Can you try to update to v1.4.4?

benalexau, 2024-09-09:
Updated and relative mode including clicking the left mouse button works reasonably well under FireFox 130.0 Linux x86_64. I didn't test other mouse buttons / wheels etc.

Unfortunately Chrome 128.0.6613.113 Linux x86_64 does not provide a stable mouse pointer in relative mode. It is erratic and unusable.

ThomasVon2021, 2024-09-09:
I use windows and mac chrome test ok. let me find a ubuntu to test.

blikvm, 2025-08-04:
In the latest version v2.0.4, there are more user tests compared to the mouse, and currently no bugs have been reported. You can try upgrading.