blikvm/blikvm #85: language setting do not get saved

devzwf, 2023-11-19:
how to reproduce : 

1. login
2. choose a different language than english
3. logout
4. login again 

language reset to english

ThomasVon2021, 2024-01-07:
It was specially designed in this way. Taking mobile phones as an example, it was found online that there are more than 100 languages built-in on the phone. Many people change a language they don't know, such as Arabic, and then become confused and unable to change it back. They can't even find the menu. Therefore, when designing for multiple languages, I assumed that multiple languages are a temporary setting that cannot be saved.

ThomasVon2021, 2024-01-07:
One idea is that the login interface provides the ability to switch languages on its own, and the UI can be designed as the flags of various countries to solve this problem.

m50S79sM6SRNp8Jn, 2024-01-07:
This behavior will be addressed in our new BliKVM Matrix release, tentatively Q1/Q2 2024.

ThomasVon2021, 2024-08-25:
https://wiki.blicube.com/blikvm/en/flashing_os/
You can flash the lasted image to test.

closed 2024-08-25 by ThomasVon2021

devzwf, 2024-08-25:
will test