blikvm/blikvm #111: Firewall issue on CM4 - http and webrtc are blocked by default

harlong, 2023-12-05:
On just flashed CM4 (image blikvm-v1-v2-v3-20231203.zip, blikvm v1.3.3) ports 80 (http) and 8188 (webrtc) were filtered by default. I had to do this to get webUI work:

sudo -i
rw
iptables -A INPUT -p tcp --dport 80 -j ACCEPT
iptables -A INPUT -p tcp --dport 8188 -j ACCEPT
iptables-save > /etc/iptables/rules.v4

It seems that /etc/iptables/rules.v4 should be corrected in image file

ThomasVon2021, 2023-12-06:
That is to say, burning the blikvm-v1-v2-v3-20231203.zip to CM4 version, the web cannot be opened? I will test it tonight to see if it works.

harlong, 2023-12-06:
Yes, it says "connection timed out"
[image]

[image]
The same time in netstat we see, that kvm-link, janus and ustreamer are listening on their ports

And in iptables -L - default policy for INPUT chain is "drop" and ports 80, 8008, 8188 are not explicitly open

ThomasVon2021, 2023-12-06:
Learned, thank you for explaining. I will update the image tonight.

harlong, 2023-12-06:
Not confirmed. In the newly flashed image everything works normal. The case was that user occasionly installed firewall by himself.

closed 2023-12-06 by harlong