blikvm/blikvm #114: [Feature Request] On screen mouse click control / Touch mode

boudrar, 2023-12-07:
In my use case, i need to use BliKVM from a smartphone.
because there is no VNC support yet, making clicks, drag/drops is impossible by touchscreen on Android via th web UI

It would be cool if there was an option to have right click, left click, middle click, right click hold/release ( for dragging) buttons
(and having a "trackpad" like [mmiscool/JS-virtual-Touchpad](https://github.com/mmiscool/JS-virtual-Touchpad) would be AWESOME!)

toggles to show/hide the buttons in the UI or make them appear automatically if a phone is detected with something like detectmobilebrowsers.com

ThomasVon2021, 2024-11-22:
[image]

I have identified this issue and it has been resolved through testing on my end. It will be released in v1.5.2

m50S79sM6SRNp8Jn, 2025-07-22:
Related to https://github.com/blikvm/blikvm/issues/288

blikvm, 2025-08-19:
https://github.com/user-attachments/assets/e0bf82ab-12d2-4009-81d4-5bdfb06ecedd

closed 2025-09-09 by blikvm