blikvm/blikvm #116: Develop 1.3.3 - 1.3.5. - web based user / password manager does not work

swingstate, 2023-12-14:
Tried with both 1.3.3 and after updating to 1.3.5 - Changes made to the admin account are not saved when pressing the "ok" button. 

Also tried `kvmd-htpasswd` in the ssh terminal, but apparently that package isn't included in the image? `-bash: kvmd-htpasswd: command not found` is what I got back. 

changing BLIKVM password in the ssh terminal worked. 

@ThomasVon2021 Can you share a working process to change the admin password?

ThomasVon2021, 2023-12-18:
Okay, I will test it. Kvmd htpasswd is definitely not available, it is not a service of blikvm　os．

ThomasVon2021, 2023-12-18:
My test is effective. After clicking OK, restart KVM, close the current web window, and open a new window.

swingstate, 2023-12-31:
Is there an option for extended logging which I could enable to trace the issue on my end?

ThomasVon2021, 2024-01-06:
> Is there an option for extended logging which I could enable to trace the issue on my end?

Upgrading to 1.3.9 should solve your problem. I have found a bug. Can you conduct upgrade testing and provide me with feedback?

swingstate, 2024-01-07:
I did the upgrade through the shell script and could change the password for admin. However, when I then clicked the "reboot KVM" button in the webinterface, the KVM didn't come back online and I had to power cycle it. Something users should be aware or this should be looked at.

I now can login with the new password.

ThomasVon2021, 2024-01-07:
Ok, if you don't update, can you click the reboot again to test? I test it, and later when I refresh Windows, it will recover.

ThomasVon2021, 2024-08-25:
https://wiki.blicube.com/blikvm/en/flashing_os/
You can flash the lasted image to test.

closed 2024-08-25 by ThomasVon2021