blikvm/blikvm #132: HDMI Switch not working with PiVKM

howetech, 2024-01-19:
I can't get the BliKVM-Switch-V1.0 working with PiKVM using the instructions from here:

https://wiki.blicube.com/blikvm/en/switch-guide/

The switch works fine using BliKVM software, but when using PiVKM, the GPIO menu buttons don't work and they don't show the green light for the current switch position.

Is there anything additional required to get the provided KVM FTDI USB adapter to work with PiVKM?

ThomasVon2021, 2024-01-23:
Can you put your config about PiKVM here. I think you didn't write it right somewhere.

howetech, 2024-01-23:
PiVKM config is exactly per the guide above.  

After restarting kvmd the following process then runs at 100% CPU:

kvmd/gpio-xh-hk4401-hk: /usr/bin/python /usr/bin/kvmd --run

My /etc/kvmd/override.yaml:

###############################################################################
#                                                                             #
#  Override KVMD settings.                                                    #
#                                                                             #
#    * https://docs.pikvm.org/first_steps/#structure-of-configuration-files   #
#                                                                             #
#  This file uses YAML syntax. All overridden parameters will be applied      #
#  AFTER other configs and "!include" directives, and BEFORE validation.      #
#  Values should be combined under common sections.                           #
#                                                                             #
###############################################################################
#
#########
# Wrong #
#########
#kvmd:
#    gpio:
#        drivers: ...
#kvmd:
#    gpio:
#        scheme: ...
#
###########
# Correct #
###########
#kvmd:
#    gpio:
#        drivers: ...
#        scheme: ...
#
###########
# Example #
###########
#vnc:
#    # See https://docs.pikvm.org/vnc
#    keymap: /usr/share/kvmd/keymaps/ru # Set russian keymap
#    auth:
#        vncauth:
#            enabled: true # Enable auth via /etc/kvmd/vncpasswd

kvmd:
    gpio:
        drivers:
            hk:
                type: xh_hk4401
                device: /dev/ttyUSB0
        scheme:
            ch0_led:
                driver: hk
                pin: 0
                mode: input
            ch1_led:
                driver: hk
                pin: 1
                mode: input
            ch2_led:
                driver: hk
                pin: 2
                mode: input
            ch3_led:
                driver: hk
                pin: 3
                mode: input
            ch0_button:
                driver: hk
                pin: 0
                mode: output
                switch: false
            ch1_button:
                driver: hk
                pin: 1
                mode: output
                switch: false
            ch2_button:
                driver: hk
                pin: 2
                mode: output
                switch: false
            ch3_button:
                driver: hk
                pin: 3
                mode: output
                switch: false
        view:
            table:
                - ["#Input 1", ch0_led, ch0_button]
                - ["#Input 2", ch1_led, ch1_button]
                - ["#Input 3", ch2_led, ch2_button]
                - ["#Input 4", ch3_led, ch3_button]

ThomasVon2021, 2024-01-24:
Are you using the pikvm image provided by blikvm? Let me see which image to use for testing.

howetech, 2024-01-24:
I am using the version from the PiVKM website as it is more up to date than the image provided by blikvm which is dated 2022.

https://files.pikvm.org/images/v3-hdmi-rpi4-box-latest.img.xz

ThomasVon2021, 2024-01-26:
So when you use the image provided by blikvm, is it okay to use this configuration because the wiki was written based on the pikvm image from that time in 2022.

howetech, 2024-01-26:
I just did a clean install with the PiKVM image provided by blikvm:

https://drive.google.com/drive/folders/1DcpxSzjbhM7wijaldql2UI4pUyEhOTCJ?usp=share_link

Exactly the same issue.  The GPIO buttons don't work and the GPIO status LED doesn't light up either.

The kvmd/gpio-xh-hk process is also running at 100% after restarting kvmd with the override settings from the blikvm website.

meszaroszoltan, 2024-02-02:
I just got mine, and I have the same issue. Configured exactly as the website instructs, but I cannot switch inputs via the software. Using the stock PiKVM image.

[image]

meszaroszoltan, 2024-02-04:
@ThomasVon2021, do you have any recommendations or pointers on how to troubleshoot this issue?

ThomasVon2021, 2024-02-05:
I think there may be an error in the format of the config file when pasting it. I will download this image soon and provide the configuration file directly.

ThomasVon2021, 2024-02-05:
> @ThomasVon2021, do you have any recommendations or pointers on how to troubleshoot this issue?

Before this, you can use the serial debugging assistant to test whether the switch can be switched based on the control protocol of the switch. If the test is conducted, please let me know the result.

ThomasVon2021, 2024-02-05:
> @ThomasVon2021, do you have any recommendations or pointers on how to troubleshoot this issue?

Also, have you checked if the/dev/ttyUSB0 device descriptor appears? If a cable with a label other than KVM is used, it will not work.

howetech, 2024-02-05:
Running the below command from the shell works to switch the KVM inputs, so the cable and connectivity is working for me, just not the PiKVM yaml config:

echo "SW1\r\nG01gA" > /dev/ttyUSB0

ThomasVon2021, 2024-02-05:
Ok, I will confirm if the format of the wiki is incorrect, and if you have time, you can also refer to the official documentation of pikvm for debugging.
https://docs.pikvm.org/gpio/#scheme

ThomasVon2021, 2024-02-05:
https://github.com/ThomasVon2021/blikvm/assets/45224459/b8db8d6c-ac8e-43f9-913c-cbadfaae84b2

[override.yaml.zip](https://github.com/ThomasVon2021/blikvm/files/14167853/override.yaml.zip)

Hello, I have provided the configuration files and test videos on my end. According to the wiki, everything is normal. You can check if there are any inconsistencies in the operation.

meszaroszoltan, 2024-02-05:
Thanks for your effort @ThomasVon2021.
I downloaded and unpacked the zip you provided and used its content as `override.yaml`, but nothing has changed :/ 

It is less than ideal, but at least I can use `echo "SW1\r\nG01gA" > /dev/ttyUSB0` for switching.

ThomasVon2021, 2024-02-06:
> investigate

Any hardware version is fine, I tested it using HAT. The image used is hat20221004

ThomasVon2021, 2024-02-06:
> @ThomasVon2021 what version of PiKVM did you test with?
> 
> One thing I've noticed is that when I apply the config and it doesn't work, the xh_hk4401 process runs at 100% CPU.
> 
> Any chance you could post your copy of this file as well:
> 
> /usr/lib/python3.11/site-packages/kvmd/plugins/ugpio/xh_hk4401.py

I download hat20221004.zip, flash to sd card, change override.yaml, and everything is normal.

ThomasVon2021, 2024-02-06:
By the way, you can do a test and open the serial assistant to see if you can receive the message sent back by the switch. A very small portion of the hardware has this defect, which can only be controlled but does not return a message, making it possible for PiKVM's web to not recognize it. In that case, I can resend a new switch to you.

meszaroszoltan, 2024-02-06:
> By the way, you can do a test and open the serial assistant to see if you can receive the message sent back by the switch. 

Excuse my ignorance, but I don't know which tool you refer to, and I couldn't track it down.
However, I did a test like this:
[image]
Does this prove your theory about the missing response? I can double check it if you tell me how to do it with serial assistant.

ThomasVon2021, 2024-02-06:
Like putty.

meszaroszoltan, 2024-02-06:
Tried it with `minicom`, but I am unable to communicate with it. 

I guess my device falls into the "can only be controlled but does not return a message" category. I will sort out the rest on AliExpress with BLI Store, thanks for the help.

ThomasVon2021, 2024-02-06:
No problem, leave me a message on Discord and I will resend it to you

meszaroszoltan, 2024-02-07:
I was mistaken. After all, I was able to communicate with the device, so it isnt a hardware issue 
[image]

iLLiac4, 2024-02-08:
Hi i have same issue. Commands work well and they are switching:
echo "SW1\r\nG01gA" > /dev/ttyUSB0
echo "SW2\r\nG02gA" > /dev/ttyUSB0
echo "SW3\r\nG03gA" > /dev/ttyUSB0
echo "SW3\r\nG03gA" > /dev/ttyUSB0

But nothing is reflected on the GUI. Pressing buttons on the gui it does not send any signal (the led does not blink on serial adapter). When entering the command over terminal it blinks and switches.

ThomasVon2021, 2024-02-08:
> Hi i have same issue. Commands work well and they are switching: echo "SW1\r\nG01gA" > /dev/ttyUSB0 echo "SW2\r\nG02gA" > /dev/ttyUSB0 echo "SW3\r\nG03gA" > /dev/ttyUSB0 echo "SW3\r\nG03gA" > [...]

Which version of PikVM image are you using? I think PikVM needs to assist in troubleshooting because the serial debugging assistant can already determine that the hardware is working properly.

iLLiac4, 2024-02-09:
Hi. I have found the fix and it works now.

Edit file /etc/kvmd/override.yaml and add the line:
protocol: 2

Sample:
---------------------------
kvmd:
    gpio:
        drivers:
            hk:
                type: xh_hk4401
                protocol: 2
                device: /dev/ttyUSB0
        scheme:
            ch0_led:
                driver: hk
------------------------------

meszaroszoltan, 2024-02-09:
I can confirm it really works, thanks @iLLiac4!

If someone stumbles upon this issue, and it doesn't work after adding `protocol 2`, then do a `pacman -Syu` and a `reboot`. After the reboot, verify the presence of the protocol config with`kvmd -m | grep protocol`.

ThomasVon2021, 2024-02-18:
Thank you all, I will add this to the wiki

closed 2024-02-18 by ThomasVon2021

Thormus, 2024-07-26:
> I can confirm it really works, thanks @iLLiac4!
> 
> If someone stumbles upon this issue, and it doesn't work after adding `protocol 2`, then do a `pacman -Syu` and a `reboot`. After the reboot, verify the presence of the protocol config with`kvmd -m | grep protocol`.

kvmd -m | grep protocol returns nothing for me.