blikvm/blikvm #134: [Enhancement] Multiple Key Capability in OSD Keyboard

owendelong, 2024-01-23:
Often, combinations of keys (e.g. Ctrl-Alt-F2) are needed when trying to interact with a remote server via the KVM.

The current OSD keyboard treats each key as pressed only so long as you hold the mouse button down on it.

To perform multi-key combinations, it is necessary to be able to mark a key to be held down. This doesn't apply to most keys, but ideally, for OSD keyboard, the following keys should be toggles rather than momentary, but they should release after being sent in combination with a key not on the list (e.g. the Ctrl and Alt toggles should be released after sending Ctrl-Alt-F2).

- Left Shift
- Left Ctrl
- Left Option/Alt
- Left CMD/Pillow/Apple
- Right Shift
- Right Ctrl
- Right Option/Alt
- Right Cmd/Pillow/Apple
- FN (if applicable)

This would greatly increase the utility of the OSD keyboard.

m50S79sM6SRNp8Jn, 2024-01-28:
Hi Owen, incidentally, for which software version and which device type did you raise this issue?

owendelong, 2024-01-28:
1.3.9

ThomasVon2021, 2024-09-20:
https://wiki.blicube.com/blikvm/en/hid/

ThomasVon2021, 2024-09-20:
https://wiki.blicube.com/blikvm/en/hid/
Can you try to update v1.4.5-alpha to test?

ThomasVon2021, 2024-11-21:
Can you try to update v1.5.1 to test? If you still have a problem, you can reopen this issue.

closed 2024-11-21 by ThomasVon2021