blikvm/blikvm #135: [Bug] A second web session gets a cryptic error message

owendelong, 2024-01-24:
Experienced on Blikvm software 1.3.9 on v4

When a second tab is connected to the KVM and logs in, the original tab will get a graphic stating that the device is off line and check network within the blikvm UI. One can still do some things if one tries hard enough, but you cannot get the target screen to work.

Instead of "Device is Offline, CHECK Network!" perhaps it would be better to say "Device is in use by another session" or something to that effect. Ideally provide the IP address of the active remote device.

owendelong, 2024-01-26:
No reverse proxies... Will see if I can figure out how to record a video.

owendelong, 2024-01-26:
I was unable to reproduce the problem, but I will try to capture more data if I am able to reproduce.

m50S79sM6SRNp8Jn, 2024-01-27:
Not able to reproduce.

closed 2024-08-25 by ThomasVon2021