blikvm/blikvm #138: Update Tailscale

swingstate, 2024-02-02:
Hi Thomas,

how can I update the Tailscale version from 1.40.1 to the latest 1.5x? Looks like is no newer package for this distro?

I would like to use the autoupdate feature which got introduced in 1.52.

Thanks

ThomasVon2021, 2024-02-04:
You can uninstall and reinstall it.

swingstate, 2024-02-04:
reinstall ends with 1.40.1

m50S79sM6SRNp8Jn, 2024-02-04:
Have you tried contacting Tailscale on the best course of action?
Please let us know what they said.

ThomasVon2021, 2024-02-06:
> Have you tried contacting Tailscale on the best course of action?
> Please let us know what they said.

What is your installation method?

ThomasVon2021, 2024-02-07:
I will close this issue. If you have successfully upgraded, you can Google search for instructions on how to uninstall and install the latest version in a Linux environment.
If there are still issues, you can reopen this issue.

closed 2024-02-07 by ThomasVon2021

masterkain, 2024-12-26:
I had to do this:

```
root@blikvm(rw):/home/blikvm# chmod 1777 /tmp
root@blikvm(rw):/home/blikvm# curl -fsSL https://tailscale.com/install.sh | sh
```

furthermore passing `--accept-routes` disables normal lan access for me, had to `tailscale up --accept-routes=false`