blikvm/blikvm #141: [BliKvm V3] Problem of using general USB msd

KaoTuz, 2024-02-05:
Hi

I have an problem of using general USB msd

I had created a usb and put some file in it by SCP command

and it could see and mount on the target side successfully

But when i want to add new file to the usb by the following two command

1. Copy file to kvm
`scp xxx blikvm@xxxx:/mnt/msd/user/`
2. Sync file and target side
`sudo bash /usr/bin/blikvm/kvmd-msd.sh -c forward`

The new file upload successful to the kvm terminal
But the new file do not sync to the target side,i could not see my new file in my target usb disk

First Transfer <--- It's OK
     I create  an folder efi/boot  and put a file shellx64.efi in path  /mnt/msd/user/efi/boot
Second Transfer  <-- Could not see my new added file
    i delete file shellx64.efi  in path  /mnt/msd/user/efi/boot, and add a file bootX64.efi into path  /mnt/msd/user/efi/boot

is there anything that i need to notice or missing?

Code Release Version : V1.4.0
the Log after executing step2
[Log.txt](https://github.com/ThomasVon2021/blikvm/files/14161544/Log.txt)

ThomasVon2021, 2024-02-05:
Can you test again without creating subdirectories and only placing files in the path? Based on the code, the script currently does not support creating folders. I want to confirm this.

KaoTuz, 2024-02-05:
I tried the following procedure and it had wired situation

1. Create 1G usb disk
    ```
   sudo bash /usr/bin/blikvm/kvmd-msd.sh -c make -s 4 -t other
    ```
2. First Transfer , put file by scp command and the file sync properly
[image]

3.  Second Transfer , Add ReadME.md file by scp command and the file sync properly
[image]

4. Third Transfer , remove ReadME.md and add LICENSE file , sync incorrectly
[image]

The deleted file(READNE.ms) sync to target magically, even the file had been removed in kvm terminal

ThomasVon2021, 2024-02-05:
This is within expectations. You see it as a USB drive, where the files are copied in and the source is deleted, but the files in the USB drive are still there.
Unless you delete it on the target computer.

ThomasVon2021, 2024-02-07:
May I ask if there are any further discoveries? If not, I plan to improve the readme. Regarding the two usage precautions you have discovered, I will then close the issue.

KaoTuz, 2024-02-07:
Is there would be an improved for creating sub directories ?

ThomasVon2021, 2024-02-28:
> Is there would be an improved for creating sub directories ?

https://github.com/ThomasVon2021/blikvm/blob/master/package/kvmd-msd/kvmd-msd.sh

Based on your requirements, we have fixed this issue. Can you manually use the latest script for testing first and then provide feedback on the results?

closed 2024-08-15 by ThomasVon2021