blikvm/blikvm #143: [Feature Request]  Implement Tesmart KVM switch 8/16 Ports on BlikvmOS

soares23, 2024-02-07:
Hi,

First thx for you great work. 
I would to ask if possible to implement the TESmart 8 and 16 port IP-KVM switches with RS232 Serial, lan console Port.

I have found on other github page link: https://github.com/bbeaudoin/bash/tree/master/tesmart information about that KVM Switch an how to communicate with that KVM Switch.

Thx in advanced if possible to integrate it. An sorry for my not so good english .

EthanSchmidtke, 2024-07-29:
Hello,

I'm not sure where on the roadmap this features falls (If anywhere) but I wanted to state that the company I work for is currently debating purchasing BliKVM V4 or PiKVMs to overhaul our testing department. We currently have 10 TESmart 16 port KVMs but in  the next few months we will be expanding to ~50 for our testing bays. If there is a plan to support the TESmart KVMs on the BliKVM I will be putting in an order to purchase one for each switch once the feature is available. Otherwise, we will likely be going to with the PiKVM which already has support (See [PiKVM Wiki](https://docs.pikvm.org/tesmart/)) by the end of the year at the latest.

To be clear, I love the BliKVM (Already bought one personally) and will support it regardless. I just wanted to give some attention to this feature as it's something I would love to see implemented and it would be awesome to support it through my work.

Thank you for the hard work and love put into this project!

ThomasVon2021, 2024-09-19:
Hi, now blikvm os support two switch, can you know about anything about nodejs.
https://github.com/ThomasVon2021/blikvm-web-server/tree/master/src/modules/kvmd/switch
Now it is easy to add other switch support.

ThomasVon2021, 2024-11-26:
Is your development going smoothly? We cannot buy this switch in our country. If you could send me a switch for testing, I would be happy to develop and support this feature for you.

soares23, 2024-11-26:
Hello, unfortunately I can't develop this because I don't know any programming language. Which country are you in? I can send you a switch so that you can implement it and as soon as you have done this you can send it back to me.

ThomasVon2021, 2024-11-27:
Thank you very much for your help. I am in China. Can you give me your email so that I can send you the address?

taylorjonl, 2024-12-26:
I just got my blikvm today and have this switch, you guys work something out?  Any progress on this?  I am a software engineer by day, nodejs isn't my primary language but I am highly motivated to get these two to work together so I don't have to keep walking downstairs to my rack to switch inputs.

ThomasVon2021, 2024-12-26:
> I just got my blikvm today and have this switch, you guys work something out? Any progress on this? I am a software engineer by day, nodejs isn't my primary language but I am highly motivated to get these two to work together so I don't have to keep walking downstairs to my rack to switch inputs.

It would be great if you could participate. I have received the switch sent by a kind-hearted user. If you are involved for any reason, I can assist you in setting up the current web development environment.

taylorjonl, 2024-12-26:
That sounds great, I just ordered a USB to 3 pin RS232 adapter so I can control it through the serial port on the back.  Tomorrow I will plug in the LAN cable and see if I can control it across IP based on that repo linked above.  Do you have any docs on setting up a development environment or if I find a standard nodejs tutorial for development, is that a good starting point?  Going to start reviewing the code you linked above concerning the existing switch integration until tomorrow.

ThomasVon2021, 2024-12-26:
https://github.com/ThomasVon2021/blikvm/blob/master/dev-readme.md
[image]

ThomasVon2021, 2024-12-26:
> That sounds great, I just ordered a USB to 3 pin RS232 adapter so I can control it through the serial port on the back. Tomorrow I will plug in the LAN cable and see if I can control it across IP [...]

You can add Discord. My name on Discord is Vons, and we can communicate on the dev channel or directly.
https://discord.gg/9Y374gUF6C

taylorjonl, 2024-12-26:
Ok, I have a direction to go.  From my initial study of the code I am going to start by integrating using the RS232 3 pin header on the back of the KVM, since the current implementation appears like it is centered around a serial integration.  It should be pretty straight forward to get that working.  Long term I may want to try getting it working also using IP but that could require a bit of refactoring/engineering to work.  It all depends on how much desire I or others have to control the KVM across IP, to make that work you have the complexity of having to make sure you have the KVM configured to be on the same subnet as the blikvm and in all honesty, a $10 USB to 3 pin adapter is pretty easy to hook up and then you shouldn't have to care about all the IP networking complexities.

ThomasVon2021, 2024-12-26:
Yes, I agree. Let's start with serial integration. By the way, which KVM hardware do you use? I think KVMs don't have RS232 pins; they only have USB or serial first.

taylorjonl, 2024-12-26:
I have the HKS1601-L23 model.  It doesn't have a real serial port, but the green pins on the back are a TxD, GND and RxD pin, so a USB to 3 pin RS232 should work.  I just went downstairs to check that I wasn't mistaken, they are for sure there and labeled correctly.  Here is the manual I referred to:

https://support.tesmart.com/hc/en-us/article_attachments/29647436434585

My biggest issue is that they are some strange connector, going to have to figure out what they are since the box for this switch is long gone so any accessories are lost to time.

BTW, I cut a branch on my fork and pushed an initial commit that likely works to control the KVM, just won't stay in sync: https://github.com/taylorjonl/blikvm-web-server/tree/feature/tesmart-switch

Now, going to get on Amazon and hope I can find this connector.

ThomasVon2021, 2024-12-26:
Ok, your are right, I thought you said blikvm has a 232 interface. 
Suggest developing in the dev branch, only need to develop the backend, we are currently refactoring the frontend.
https://github.com/ThomasVon2021/blikvm-web-server/tree/dev

ThomasVon2021, 2024-12-26:
https://github.com/ThomasVon2021/blikvm-web-server/blob/dev/config/switch.json
The switch control file has been separated separately.

taylorjonl, 2024-12-26:
Ok, I see that, I will clean it up tomorrow.  The connector is called a phoenix connector, unfortunately I can't get one until Saturday, unless I can find one randomly in my mess of random stuff I have.  I had a project related to some lighting that did RS232 that may have this connector, will check tomorrow.  Keep your fingers crossed, otherwise I will have to wait until after Saturday to really progress much since my next step is to test if I can control the KVM and next would be to see what kind of data it feeds back across the serial connection when you change the input on the actual KVM from the buttons.

ThomasVon2021, 2024-12-26:
https://github.com/pikvm/kvmd/blob/master/kvmd/plugins/ugpio/tesmart.py

taylorjonl, 2024-12-26:
Looking at this line: https://github.com/pikvm/kvmd/blob/647d3f396117b431bead5b4ae47cf53fc546d96f/kvmd/plugins/ugpio/tesmart.py#L140

I think my code will actually work to switch the KVM, the biggest gap I have at the moment is keeping blikvm in sync with the KVM, in case the user is making manual switches using the buttons on the front.

ThomasVon2021, 2024-12-26:
Yes, let's see if there are any commands that can read the current channel.

taylorjonl, 2024-12-26:
There are commands that can be used to get the current channel but I am trying to model the same integration you have for the other switch where it expects the serial connection to receive data(an event) when the channel changes.  At this point I am stuck until I can figure out how to connect to this port.  At this point I am slightly worried it may take a while to get a connector, I thought it was a pheonix connector but I am beginning to doubt it.  This is the connector:
[image]
Not sure if someone has seen or can identify this connector.

I did find an RS232 USB adapter with individual leads lying around, my issue now is figuring out what each color wire is since these are unlabeled and I don't know where I got these from, so today I hope to identify the wire colors and possibly get some connectivity to the KVM to see if manual changes on the front result in a data packet being sent along the serial cable.

HarrisonWilson, 2024-12-27:
https://support.tesmart.com/hc/en-us/article_attachments/27716605047961

Serial commands for reference

taylorjonl, 2024-12-27:
> https://support.tesmart.com/hc/en-us/article_attachments/27716605047961
> 
> Serial commands for reference

Thanks, I had already found that.

I spent yesterday struggling to verify that the KVM functions across the RS232 port, it was quite frustrating.  Partially because I don't have the needed hardware(connectors, jumpers, etc) to easily connect to the port.  All I had available to me was a USB to 4 wire RS232 adapter and an Arduino.  Either one or both of those were bad or something, so I went to bed last night with no progress.  Because Amazon has a facility close I ordered a few more RS232 adapters which got here this morning.  Not sure if it was the night sleep or these new adapters work better but I now have basic control over the KVM using a simple C# application I wrote to send commands to it.

My biggest issue is that I don't get any data back from the KVM, I can switch inputs, but I have no way to query what the current input is.  According to their docs, the PIKVM code and the other documentation linked above, I am sending the correct command but I never get any data back on the RS232 port.

So I decided to download the manufacturers application.  With that application, when I use the LAN it can keep in sync with the current input, but the serial controls don't seem to have the ability to query the current input either.  I can switch ports with it, but it can't keep in sync with the current active port if I manually change the port.  I am reaching out to the switch manufacturer now to see if they have some feedback.

My next experiment is going to be trying to sniff what their application is sending across the RS232 port.  It is quite possible my KVM has a faulty RXD line on the RS232.  Who knows.

Tomorrow I think I get my 3 pin connector which should make this easier for me, at the moment connecting to the pins is a challenge.  I am trying to verify all the functionality of the KVM before I try to get the blikvm working.

taylorjonl, 2024-12-28:
I believe my KVM switch has a faulty RX pin on the serial console, I don't get any data back from it.  If I take my RS232 adapter, cross the RX/TX pins, then connect to the port with their software, I see it issue this request:

0xAA 0xBB 0x03 0x10 0x00 0xEE

This request is used to query the active port.  That same software issues the same request for the LAN side to keep it in sync.  Which makes me believe my KVM switch has a faulty RX pin.

Is someone with one of these units and a 3 wire RS232 adapter willing to run a test for me?  I would like to see what comes back on the serial port after issuing this command.  You can run this C# code to issue this request:

```
            using SerialPort serialPort = new SerialPort("COM25", 9600, Parity.None, 8, StopBits.One);
            serialPort.Handshake = Handshake.None;
            serialPort.Open();

            serialPort.DataReceived += static (sender, _) =>
            {
                SerialPort serialPort = (SerialPort)sender;
                byte[] buffer = new byte[6];
                for (int i = 0, c = 0; i < 6; i += c)
                {
                    c = serialPort.Read(buffer, i, 6 - i);
                }
                Console.WriteLine(string.Join(" ", buffer.Select(x => $"0x{x:X2}")));
            };

            serialPort.Write(new byte[] { 0xAA, 0xBB, 0x03, 0x10, 0x00, 0xEE }, 0, 6);

            Console.WriteLine("Press ENTER to exit...");
            Console.ReadLine();
```

Just change the COM25 value to the correct COM port value.  The PiKVM code appears to expect a response of 6 bytes, I wouldn't be surprised to see 8 bytes, because when using the LAN interface it returns this response(found using wireshark):

0x02 0x18 0xAA 0xBB 0x03 0x11 0x02 0x18

When the active port is #3.

Also, I would love to know if the serial port receives data when you manually change the input on the front, the LAN side sends a packet when the inputs are changed.  You should be able to use this C# code to monitor the port:

```
            using SerialPort serialPort = new SerialPort("COM25", 9600, Parity.None, 8, StopBits.One);
            serialPort.Handshake = Handshake.None;
            serialPort.Open();

            serialPort.DataReceived += static (sender, _) =>
            {
                SerialPort serialPort = (SerialPort)sender;
                byte[] buffer = new byte[6];
                for (int i = 0, c = 0; i < 6; i += c)
                {
                    c = serialPort.Read(buffer, i, 6 - i);
                }
                Console.WriteLine(string.Join(" ", buffer.Select(x => $"0x{x:X2}")));
            };

            Console.WriteLine("Press ENTER to exit...");
            Console.ReadLine();
```

Again, change the COM port value, run the program, then change the ports on the front of the KVM.

I sent an email to Tesmart to see what they say about my unit not receiving any data.  Not sure they will help me out with a new unit, otherwise I am not sure what I am going to do.  Not sure the wife will be ok with my spending another $200+ so I don't have to walk up and down the stairs.

ThomasVon2021, 2025-04-09:
[image]
Use LAN port, why send and receive the same value? And not get active channel value.

ThomasVon2021, 2025-04-22:
V1.5.9 version adds Tesmart KVM switch control. Is there anyone who can test?

Simon-CR, 2025-09-02:
is there a plan to move this forward and support IP based switching as well ?

m50S79sM6SRNp8Jn, 2025-10-28:
> ...maybe it would be better to have some kind of config-driven thing that lets users configure a script to call that will do whatever they need it to do? e.g. "switch to channel x" might get translated to `/path/to/my/switcher_script.sh --channel x`.

Yes I agree model driven approach is needed for self service reasons.
Let us get current planned work out of the way and then we get to this one.

blikvm, 2026-10-28:
Can you try to update the v3.0.1 image. It support this function now. https://github.com/blikvm/blikvm/releases
Much better than v2.x.x version. You can also play first https://blikvm.com/demo/

acook8103, 2026-10-28:
Of course just yesterday I got ChatGPT's help to create a virtual socket to connect to my TESMART.

The demo doesn't let us disable the Switch, which I'm assuming is why I cannot change the model or otherwise inspect how to include my own models? Does v3 require reflashing the SD card?

blikvm, 2026-10-28:
Yes, v3 need to reflash the SD card

acook8103, 2026-10-30:
I'm running 3.0.1 on a V4. I was hoping network Tesmart was available. I don't have USB, so I returned to ChatGPT to reimplement a virtual serial device. I found System Scripts and the virtual device is created when I run the script manually. To make it "persistent" I tried the Run Once at startup in System Scripts, but that doesn't appear to work, even after a full reboot.

I just found the checking the recently updated documentation, I found that both /usr/libexec/blikvm-script-supervisor and /etc/init.d/S98blikvm-scripts are missing from my installation.

Is this functionality expected to work in OS 3.0.1, or is it still in development?

Also, Is this even the expected way to add custom KVMs moving forward?

blikvm, 2026-10-30:
The use of custom scripts will be supported in the next version.

blikvm, 2026-11-05:
> 29\. 我用的是 V4 系统，运行的是 3.0.1 版本。我原本希望网络 Tesmart 功能可用。由于没有 USB 接口，我只能回到 ChatGPT 重新实现一个虚拟串口设备。我找到了系统脚本，手动运行脚本后虚拟设备就能创建出来。为了让它“持久化”，我尝试在系统脚本中设置“启动时运行一次”，但似乎不起作用，即使重启系统后也一样。
>      我刚刚查看了最近更新的文档，发现我的安装中缺少 /usr/libexec/blikvm-script-supervisor 和 /etc/init.d/S98blikvm-scripts 这两个目录。
>      此功能预计在 OS 3.0.1 中可用，还是仍在开发中？
>      另外，这是否是未来添加自定义 KVM 的预期方式？

Can you try to use 3.0.3 to test again.