blikvm/blikvm #144: Janus uses 100% CPU

phildabill, 2024-02-16:
On the last 2 releases, janus uses 100% cpu on a blikvm V1.

PID USER      PR  NI    VIRT    RES    SHR S  %CPU  %MEM     TIME+ COMMAND
288 root      20   0 1571972  20020  17352 S 200.3   1.1   6:08.04 janus

blikvm@blikvm(ro):~$ ps -To pcpu,tid,comm -C janus | sort -r -k1
%CPU     TID COMMAND
71.2     556 mixer 1234
71.0     552 mp 2
 0.0     586 ws thread
 0.0     574 MHD-single
 0.0     573 http timer

What is mp 2?

ThomasVon2021, 2024-02-18:
Yes, I also noticed this phenomenon. After the device runs for a few minutes, the CPU usage rate will decrease. Since I generated this code directly using open-source compilation, I have not yet researched why the CPU usage rate in the first few minutes would reach 100%

m50S79sM6SRNp8Jn, 2025-03-29:
Recommend reflashing latest BliKVM image
https://wiki.blicube.com/blikvm/en/flashing_os/

m50S79sM6SRNp8Jn, 2025-04-07:
If there is any update, please feel free to open this issue.
For now I am closing it.

closed 2025-04-07 by m50S79sM6SRNp8Jn