blikvm/blikvm #145: [Feature request] Make KVM switch channel labels customizable

mluna0101, 2024-02-20:
In the web interface it would be nice to be able to change the channel button labels to something that makes sense to what is connected to the port instead of just channel numbers.

Say, Channel 1 could be renamed to something like "Main server", Channel 2 to "NAS", Channel 3 to "Hypervisor" and Channel 4 to "Linux".

Maybe it doesn't look too useful but when you escalate it to 8 channels or more then you'll really appreciate that feature, especially if you don't access the KVM that often.

ThomasVon2021, 2024-02-20:
That's a good suggestion, we will add this feature in future versions.

ThomasVon2021, 2024-08-25:
https://wiki.blicube.com/blikvm/en/flashing_os/
You can flash the lasted image to test.

ThomasVon2021, 2024-09-09:
The latest v1.4.4 version has been tested and is now supported. If there are any issues, you can reopen it.

closed 2024-09-09 by ThomasVon2021