blikvm/blikvm #147: [documentation incorrect]Cannot set static IP as per doc, as file is not existing, doc needs update

homeelab, 2024-02-24:
Hi,

Baught recently the v4 AllWinner version. It came with Develop-1.3.8 code version. Updated it to 1.4.0

Trying to set static IP accordingly to doc located here : https://wiki.blicube.com/blikvm/en/static_ip/
Doc is saying to edit the /etc/dhcpcd.conf, but this file is not existing.
Please adapt the doc and give info on how to set a static IP properly.
I prefer to wait for feedback before attempting something wrong and being ejected from the network
Thanks.

PS: good product !  ;)

ThomasVon2021, 2024-02-25:
I will conduct testing, and I remember that this file does not exist. Just create it directly. The problem is actually how to set the static IP in the Armbian system.

ThomasVon2021, 2024-02-28:
https://github.com/ThomasVon2021/blikvm-site/commit/62076ba58a0bc9e0805752a1b021d9b38e7afc8d

Add how to set static ip on armbian. You can check wiki or this commit.

closed 2024-02-28 by ThomasVon2021

homeelab, 2024-02-28:
I consulted the wiki : THANKS for the update !  appreciated.