blikvm/blikvm #148: [documentation incorrect]Command ro/rw are not existing, device already in read write mode, doc needs update

homeelab, 2024-02-24:
Hi,
Got Develop-1.3.8 new device. Updated it to v1.4.0.
Doc is referencing "ro" and "rw" command for passing from read only to read write, but the current version, under root user, is not presenting such commands, which makes doc confusing for a new user.
such ro/rw command are not existing under the blikvm user too.
In fact, I can see that doing a "touch file" is working ok, and I can delete it too, so it is already read write enabled.
Please correct the doc to reflect usage or no usage of such commands, as maybe in v4 hardware or in v1.3.8+ there is no more need of using such commands.
But I like the idea of locking the device for data protection in case of power outage, as I know raspberry is very ticky on this and can easily get corrupted... 
Thanks.
Regards.

ThomasVon2021, 2024-02-25:
Do you mean this file:
https://wiki.blicube.com/blikvm/en/update/#manually-run-the-script-to-update

homeelab, 2024-02-27:
yes. "In the terminal，If the current system terminal can see the **ro** keyword and it is a read-only system, it is necessary to use '**rw**' to make the system writable."

ThomasVon2021, 2024-02-28:
Thank you very much for your suggestion. I will add your description to it.

ThomasVon2021, 2024-02-28:
> In the terminal，If the current system terminal can see the **ro** keyword and it is a read-only system, it is necessary to use '**rw**' to make the system writable.

You are using v4, and the terminal should not be able to see the keyword for `ro`, so there is no need to enter the rw command to change it. Do you still need to use the `rw` command if this sentence makes you misunderstand?

closed 2024-04-01 by ThomasVon2021