blikvm/blikvm #151: Login error over cloudflare tunnel

bauerhunter11, 2024-02-28:
I am experiencing an issue where I cannot login when routing the BliKVM over a cloudflared tunnel. I can view the login screen but I cannot log in. When I enter username and password and click 'login' it appears to get stuck. 

[image]

The password disappears and the login button greys out and cannot be clicked again. 

Note that this issue is not present when accessing the BliKVM locally on my network. 

Here are my cloudflare tunnel settings: 
[image]

Final note: I have the tunnel set to HTTP not HTTPS as the tunnel's connection seems to fail when I choose HTTPS, even with TLS No Verify. 

Device: v4 Allwinner

BliKVM OS: 1.4.0

bauerhunter11, 2024-03-07:
might this have to do with HTTPS?

m50S79sM6SRNp8Jn, 2024-03-10:
Yes, we will combine it.

ThomasVon2021, 2025-02-07:
https://wiki.blicube.com/blikvm/en/cloudflare-tunnels/

Can you update to v1.5.6 to test? If there are still issues after testing, you can open it again.

closed 2025-02-07 by ThomasVon2021