blikvm/blikvm #154: Blikvm not boot to OS :

irish1986, 2024-03-11:
I am unable to get my blikvm v3 HAT to work.  I have tried both version of the OS and it always fails (get stuck here). I tried this RPi4 ith Ubuntu server and it booted well.  Any suggestion ?

[image]

m50S79sM6SRNp8Jn, 2024-03-11:
Can you elaborate on the two versions, which versions specifically?

m50S79sM6SRNp8Jn, 2024-03-11:
How you tried reflashing the SD card?

m50S79sM6SRNp8Jn, 2024-03-12:
Can you try with SD card, use the latest image.

ThomasVon2021, 2024-03-12:
> I am using a USB key instead of SD card (not sure if this could be the issue). I got this RPi4 and HAT working a few months ago but decided to do a clean install to document better the steps for my lab. I am fairly certain I used a USB device since that how I boot all my RPi4.
> 
> I use both version : blikvm-v1-v2-v3-20230924 and blikvm-v1-v2-v3-20231203

The SD card startup is not related to the hat itself, but only depends on the pi4b and SD card

ThomasVon2021, 2024-04-01:
If it has not been resolved yet, please open this issue again. And provide more information.

closed 2024-04-01 by ThomasVon2021