blikvm/blikvm #155: WebUI keep resizing the video stream

ds2812, 2024-03-15:
The WebUI keep resizing the video stream from the client according to the browser size.
Could you make an option to keep the video stream fixed at ustream output resolution (1080p by default)?

m50S79sM6SRNp8Jn, 2024-03-16:
Which software version you are using?
Which video type, h264 or mjpeg?
Which BliKVM device did are buy?

ds2812, 2024-03-16:
Which software version you are using? - 2.2 (I've switched to piKVM now, cannot verify)
Which video type, h264 or mjpeg? - MJPEG
Which BliKVM device did you buy? - BLIKVM V4

ThomasVon2021, 2024-09-08:
You can try to update 1.4.3 alpha to test.

ThomasVon2021, 2024-09-23:
You can update to version v1.4.7 alpha for testing now. If there are still issues, you can reopen this problem

closed 2024-09-23 by ThomasVon2021