blikvm/blikvm #156: Switch off USB-PC2?

bunducafe, 2024-03-19:
Is there an easy way to switch off the USB feature of the bliKVM? I know there are possibilities to do that on a pi. But usually it means that you lose also the LAN port.

Background: I am powering the pi via battery so "every watt counts", so I would love to have a decent way to switch off unneeded things and just turn them on when there is a use case.

bunducafe, 2024-03-19:
I do have the bliKVM PCIe version with a CM4 module on top of it. I am referring to the usb ports other than the power plug (I think in the documentation it states USB PC2)

I know that I will lose keyboard and mouse feature but it would be handy to activate this just when needed more actions to be done via SSH.

ThomasVon2021, 2024-03-19:
If it's just disabling the simulation function of the mouse and keyboard, I don't think it will have much impact on power consumption? Are you thinking of not starting this USB interface when the system starts? By the way, are you currently using pikvm os or blikvm os?

ThomasVon2021, 2024-03-19:
> I do have the bliKVM PCIe version with a CM4 module on top of it. I am referring to the usb ports other than the power plug (I think in the documentation it states USB PC2)
>
> I know that I will lose [...]

[image]

ThomasVon2021, 2024-03-19:
[image]

bunducafe, 2024-03-19:
Thanks for the quick response. Well, I ended up unplugging the USB connector of the bliKVM (that goes into the motherboard) and that is indeed working. But then - as stated - there is no possibility to have any input device if needed).

The bliKVM came preconfigured but as far as I am seeing it the regular PI-KVM OS is being used for the device.

ThomasVon2021, 2024-03-19:
If the software is enabled or not, you can refer to the official documentation of CM4 for configuration, so that the USB port will not work. You need to try it out.

bunducafe, 2024-03-19:
Thank you Thomas. I will dig into that.

Is there any benefit of using the blikvm os instead of pikvm?

ThomasVon2021, 2024-03-19:
At present, it is recommended that you use pikvm os, as blikvm os is still under continuous optimization.

m50S79sM6SRNp8Jn, 2024-03-19:
@bunducafe if you are OK with the answer can you close this issue?

bunducafe, 2024-03-20:
Yes, topic can be closed here... If you want to disable usb devices such as keyboard, mouse and mass storage then you should have a look into the dynamic usb configuration of the piKVM:
[https://docs.pikvm.org/usb_dynamic](https://docs.pikvm.org/usb_dynamic/)

closed 2024-03-20 by bunducafe