blikvm/blikvm #157: Feature request for OLED display

alandjbaker, 2024-03-20:
It would be useful if on the oled display it showed both the network ip of zero tier and the local lan. 
There is space at the bottom of the oled display for a second IP.

m50S79sM6SRNp8Jn, 2024-03-21:
For vue3 
npm install @casl/vue @casl/ability

I am starting to play with this.

m50S79sM6SRNp8Jn, 2024-03-21:
Which BliKVM device are you using?

alandjbaker, 2024-03-21:
> Which BliKVM device are you using?

The Allwinner V4

alandjbaker, 2024-03-21:
My BLIKVM is on its way three thousand miles away so I can support remote users so it would be useful for the end user to see both IP's (this would also indicate the KVM is online).

m50S79sM6SRNp8Jn, 2024-03-21:
Nice, thanks for providing the rational.

m50S79sM6SRNp8Jn, 2024-03-21:
Do you by any chance have other BliKVM's (which, quanity)?

alandjbaker, 2024-03-21:
> Do you by any chance have other BliKVM's (which, quanity)?

Just the one I'm afraid, bought from AliExpress, delivered to the UK, configured it with zerotier and dispatched to UAE!

m50S79sM6SRNp8Jn, 2024-11-22:
> It would be useful if on the oled display it showed both the network ip of zero tier and the local lan. There is space at the bottom of the oled display for a second IP.

Would it more or less useful to you having both the local LAN and VPN IP in the GUI, than in the BliKVM v4 Allwinner display?
Maybe we do both, so in the GUI and in the display, just want to get your feedback first please.

m50S79sM6SRNp8Jn, 2024-11-22:
> In my use case it would be no use in the GUI as if I'm in the GUI then I've successfully connected to it.

o/c that makes sense, we'll do both (display + GUI)

alandjbaker, 2024-11-25:
Yeah, to have this information in the GUI is surely going to be useful to some.

m50S79sM6SRNp8Jn, 2025-05-17:
Feature added. Update to latest BliKVM

closed 2025-05-17 by m50S79sM6SRNp8Jn