blikvm/blikvm #161: Blikvm v4 - No Video

jgitperth, 2024-04-05:
I have an odd issue with my BliKVM v4 AllWinner and not being able to always display video.  I've connected it to 3 PC's I have.  For 2 of them it works fine - but for the third - which is my homelab server and the reason I bought this, it wont show the video.  If I remove the HDMI cable from the Blikvm and connect it to a monitor, video displays fine, but plugged into the BliKVM I dont get video.  This machine is running an older Asrock E3-V5WS MB, Xeon processor and an NVIDIA GT 710 graphics card (if that matters!)

I see this in the logs - note the "resource temporarily unavailable" in the logs below.

 INFO  [2115.649      main] -- Starting PiKVM uStreamer 5.38 ...
-- INFO  [2115.649      main] -- Using internal blank placeholder
-- INFO  [2115.650      main] -- Listening HTTP on [0.0.0.0]:8008
-- INFO  [2115.651    stream] -- Using V4L2 device: /dev/video1
-- INFO  [2115.651    stream] -- Using desired FPS: 10
-- INFO  [2115.651      http] -- Starting HTTP eventloop ...
================================================================================
-- INFO  [2115.726    stream] -- Device fd=8 opened
-- INFO  [2115.726    stream] -- Using input channel: 0
-- INFO  [2115.747    stream] -- Using resolution: 640x480
-- INFO  [2115.747    stream] -- Using format: MJPEG
-- INFO  [2115.768    stream] -- Using HW FPS: 10
-- ERROR [2115.768    stream] -- Device doesn't support setting of HW encoding quality parameters
-- INFO  [2115.768    stream] -- Using IO method: MMAP
-- INFO  [2115.770    stream] -- Requested 5 device buffers, got 5
-- INFO  [2115.779    stream] -- Capturing started
-- INFO  [2115.779    stream] -- Switching to HW encoder: the input is (M)JPEG ...
-- INFO  [2115.779    stream] -- Using JPEG quality: encoder default
-- INFO  [2115.779    stream] -- Creating pool JPEG with 1 workers ...
-- INFO  [2115.779    stream] -- Capturing ...
-- INFO  [2119.259      http] -- HTTP: NEW client (now=1): [10.1.5.27]:57689, id=edb8670d6863c820
-- INFO  [2266.140      http] -- HTTP: DEL client (now=0): [10.1.5.27]:57689, id=edb8670d6863c820, Resource temporarily unavailable (reading,eof)

Any ideas on where to look - what to try next?
Thanks.

ThomasVon2021, 2024-04-07:
Can you try to add an HDMI lock screen treasure to test it. 
[image]

ThomasVon2021, 2024-04-07:
> Resource temporarily unavailable

Suspected to be resolution or v4 default edid, your device is not recognized

jgitperth, 2024-04-07:
@ThomasVon2021 - thanks I have ordered an EDID emulator from Amazon - will report back in a few days.

jgitperth, 2024-04-11:
It arrived today - plugged into the back of the blikvm inline with the HDMI in - and its solved my issue.

Many thanks.

ThomasVon2021, 2024-04-12:
It seems that your device doesn't recognize Lock Screen Treasure's EDD. Is there a way to force HDMI output through system settings?

closed 2024-08-25 by ThomasVon2021

banhuk, 2025-01-26:
I've tried many edid emulators over the last year and many different graphics cards.  The video display on my blikvm still only works intermittently.  Very frustrating.  This is on a dell T7910 server with dual xeons, 256gb ram running unraid.  I'd appreciate a fix as it's annoying working blind each reboot.  Thanks for listening

ThomasVon2021, 2025-01-26:
Is there a video of the loop port？
It should be noted that when the KVM web cannot retrieve the image, whether the hardware HDMI IN indicator light is on can also be determined by connecting the HDMI OUT to the monitor.

banhuk, 2025-02-02:
I just bought a NANOKVM and I no longer have video issues.  I will try the BLIKVM V4 in my other server and hopefully all will be good.  I'll report back here in due course.