blikvm/blikvm #164: Blikvm v1 skipping keypresses

vladie, 2024-05-10:
I have blikvm v1 connected to linux server
All works fine except that  it skips keypresses with some probability
For example if i enter 'hello world' in linux console (through web interface of blikvm, i'm using google chrome as client)
i can see 'helo wrld' entered or 'hell wold'
If i typing faster - probability of keypress skipping is increasing

ThomasVon2021, 2024-05-10:
May I ask if you are using blikvm ui or pikvm ui?

vladie, 2024-05-11:
blikvm ui 1.2.2

vladie, 2024-05-11:
just updated to 1.4.0 but problem still persists

ThomasVon2021, 2024-05-14:
Is it under the local area network or the public network? Is the delay significant? Are you familiar with using a browser to view messages? Every time you press a button, the browser can view it through the viewing tool.

vladie, 2024-05-14:
Local area network, no delay at all.  Yes i can view browser log but i can't find anything related to keypresses in that log.

ThomasVon2021, 2024-05-14:
It's not a log, it's a message sent by the browser. I can provide you with a screenshot when I have a device. The log is not open normally.
Of course, if you have an additional SD card, you can burn a pikvm image and test it to confirm whether it is a hardware or software issue.

vladie, 2024-05-14:
you can send me instructions what debug information you need, i will do it. my cm4 have embedded disk, no sd cards. but i can ssh to it and install package or change config values.

ThomasVon2021, 2024-05-14:
Browser viewing tool, view messages in network options.
Flash the pikvm os you can see this guide:
https://wiki.blicube.com/blikvm/en/flashing_os/#download-the-image

vladie, 2024-05-14:
where this 'messages' in firefox ? i opened developer tools -> network  and see http requests, but no messages mentioned

ThomasVon2021, 2024-05-14:
[image]
I use the chorme

vladie, 2024-05-14:
[image]

i've attached screenshot
i typed 'workwork', but wokwok appeared in output

ThomasVon2021, 2024-05-14:
After reviewing the relevant code, it was found that the two commands are within the same polling cycle, and the later ones will overwrite the previous ones. A fast typing speed can cause this problem.

vladie, 2024-05-14:
can you please fix this in new version ?

ThomasVon2021, 2024-05-14:
If you directly connect the physical keyboard to the controlled computer with a faster printing speed, see if it will be lost.

vladie, 2024-05-14:
i've setup OS on this computer without blikvm - by connecting physical keyboard and monitor - without any issues - no keypresses are lost

ThomasVon2021, 2024-05-14:
Do another test, use the assignment and paste function, click on the keyboard logo in the upper left corner, and see if it will be lost.

vladie, 2024-05-14:
Works without losing keypresses, but it sends keypresses not very fast - one by one

ThomasVon2021, 2024-05-14:
Ok，can you connect KVM to a desktop system like Windows for testing and see if it will be lost. I see if all systems have issues. I tested and controlled the Windwos system on my end, and even though I pressed it quickly, there were no issues.

vladie, 2024-05-14:
my computer runs MacOS, maybe this is different from yours ? maybe browser keypresses works differently in different OSes. i've tried firefox and chrome today (on mac) and both have lost keypresses problem

ThomasVon2021, 2024-05-14:
It refers to the controlled computer connected to KVM. My computer is also macos

ThomasVon2021, 2024-05-15:
> my computer runs MacOS, maybe this is different from yours ? maybe browser keypresses works differently in different OSes. i've tried firefox and chrome today (on mac) and both have lost keypresses problem

How is the testing situation?

vladie, 2024-05-15:
Will try to setup intel nuc today/tomorrow, connect blikvm to it and report results, i have no other computers now to test with.

vladie, 2024-05-16:
just finished setup ubuntu 24.04 server to intel nuc mini pc. results are the same as with previous computer - it skipping keypresses in console.

ThomasVon2021, 2024-05-17:
I checked the code again for any suspicious areas, but I'm not sure. Thank you very much for your cooperation. I plan to provide you with a specially validated version. Do you have time to cooperate with the testing.

vladie, 2024-05-17:
yes of course, waiting for the fix

ThomasVon2021, 2024-05-21:
[image]
After several days of investigation, I speculate that it is an issue with the USB polling rate of the controlled machine. The regular training frequency is 125Hz, which means there is an interval of 4ms per second. I have forcibly added a setting in the software that can modify this interval. The usage method is to click on the logo 10 times in a row, open the hidden project menu, set a reasonable value for the second item, such as 10, and then log in to operate. Can be tested sequentially:
1, 2, 4, 6, 8, 10, 12, 15.
Take a look at the effect
[kvm-link.zip](https://github.com/ThomasVon2021/blikvm/files/15391445/kvm-link.zip)
I have uploaded the kvm-link.zip file. After downloading, unzip it to obtain the kvm-link. Replace this file with the kvm link in the/usr/bin/blikvm/path, and then restart the system to run.

m50S79sM6SRNp8Jn, 2024-09-20:
Suggest upgrading to v1.4.5-alpha. Please let us know if the problem persists, or has since been resolved.

vladie, 2024-09-20:
root@blikvm(rw):/opt/bin/blikvm# python3 /opt/bin/blikvm/script/update.py alpha
Welcome to use the upgrade script. Please confirm that you have used git related commands before upgrading the script to ensure that update.by is in the latest state.
Board type: BoardType.V3_PCIE
The latest release tag for blikvm is  v1.4.6
The local version is  v1.4.0
Upgrading  v1.4.0  ==>  v1.4.6
Download package:  release-alpha.tar.gz  please wait...
Download progress: 1.00%
......
Download progress: 99.00%
Download progress: 100.00%
release-alpha.tar.gz downloaded to /tmp/kvm_update/release-alpha.tar.gz successfully.
Download release package success, start to install, please wait 60s...
cp: cannot stat '/mnt/exec/release/config': No such file or directory
Traceback (most recent call last):
  File "/tmp/kvm_update/release/install_release.py", line 155, in <module>
    main()
  File "/tmp/kvm_update/release/install_release.py", line 143, in main
    subprocess.check_output(cmd, shell = True, cwd=gArgs.releasepath )
  File "/usr/lib/python3.9/subprocess.py", line 424, in check_output
    return run(*popenargs, stdout=PIPE, timeout=timeout, check=True,
  File "/usr/lib/python3.9/subprocess.py", line 528, in run
    raise CalledProcessError(retcode, process.args,
subprocess.CalledProcessError: Command 'cp -R /mnt/exec/release/config /tmp && systemctl disable kvmd-janus && systemctl disable kvmd-hid && systemctl disable kvmd-main         && systemctl disable kvmd-video && bash install-kvmd-web.sh && cp package.json /usr/bin/blikvm/package.json' returned non-zero exit status 1.
Traceback (most recent call last):
  File "/opt/bin/blikvm/script/update.py", line 222, in <module>
    main()
  File "/opt/bin/blikvm/script/update.py", line 202, in main
    output = subprocess.check_output(cmd, shell = True, cwd=install_path) 
  File "/usr/lib/python3.9/subprocess.py", line 424, in check_output
    return run(*popenargs, stdout=PIPE, timeout=timeout, check=True,
  File "/usr/lib/python3.9/subprocess.py", line 528, in run
    raise CalledProcessError(retcode, process.args,
subprocess.CalledProcessError: Command 'python3 install_release.py --alpha=true' returned non-zero exit status 1.

vladie, 2024-09-20:
tried to install by directly downloading release from github, but it requires some /mnt/exec which i doesn't have

vladie, 2024-09-20:
installed successfully
needed to run ./install-kvmd-web.sh 
before python3 install_release.py --releasepath=./
which is not mentioned in your docs

vladie, 2024-09-20:
now web server is not working so i cannot test anything :(

ThomasVon2021, 2024-09-20:
I just updated the update.by script. Please update it before taking a look. The above method did not fully update and the update was successful

vladie, 2024-09-20:
i've already updated manually
bug still persist :(

vladie, 2024-09-20:
but overall UI experience much better than in 1.4.0 !!!

ThomasVon2021, 2024-09-20:
If you have time, I may need to remotely debug with you in order to troubleshoot

kiler129, 2024-11-01:
I think this isn't only a problem with v1. I have v4 and it is seriously unusable when connected to a AMD 5900x native USB port, both when running in pre-boot environments (e.g. BIOS) and after booting linux. Keys and key-combos (e.g. Ctrl+C) are being stuck CONSTANTLY and after 30 attempts to type `ls /mnt` I gave up.

m50S79sM6SRNp8Jn, 2024-11-01:
Hi @kiler129, we will be done some work on keyboard code and will be taking a closer look to have this problem resolved.

Once there is a significant update, will report back.

m50S79sM6SRNp8Jn, 2024-11-01:
> Keys and key-combos (e.g. Ctrl+C) are being stuck CONSTANTLY and after 30 attempts to type `ls /mnt` I gave up as I cannot type it without keys being lost or stuck.

When you have a moment please let me know which BliKVM version you are running.
Latest is v1.4.9-alpha (https://discord.com/channels/943534043515977768/1097914476491198554/1300539244502192219)

kiler129, 2024-11-01:
> When you have a moment please let me know which BliKVM version you are running. Latest is v1.4.9-alpha (https://discord.com/channels/943534043515977768/1097914476491198554/1300539244502192219)

Currently `v1.4.8` (on `BoardType.V4_H616`, per `git status` it's on commit `9f598f8c2f0eefa520c54386e2ace5bd19288b9d`), updated about ~1.5 weeks ago. Let me grab the 1.4.9-alpha and report.

vladie, 2024-11-14:
just updated to 1.5.0-alpha and seems that problem disappeared
but there are new bug - sometimes keyboard input stops working corretly and i need to close browser tab with blikvm and open it again to make it work

vladie, 2024-11-14:
+ not keyboard not mouse now doesn't work in bios completely :(

ThomasVon2021, 2024-11-14:
> * not keyboard not mouse now doesn't work in bios completely :(
Is this situation stable and reproducible? Is your accused computer without a desktop operating system or with a desktop operating system? Did you use MSD during this process?

vladie, 2024-11-16:
i forgot to mention that i've changed motherboard to asrock trx50 (from asus x690-e proart) and this is solved problem with disappearing keystrokes - so, this is some incompatibility with usb controller from amd x690 chipset.

m50S79sM6SRNp8Jn, 2024-11-26:
I will close this issue. If you feel it should be reopened you can do so.

closed 2024-11-26 by m50S79sM6SRNp8Jn