blikvm/blikvm #165: Make it act as a bridge

Altares, 2024-05-10:
I love the bliKVM for working on servers and PCs that are inaccessible or to far from a screen. 
I use it as a practical way to attach a KVM to a PC while I'm working on it (open chassis on my workbench).
So, while I can wirelessly access the display and input keyboard actions using my laptop. I wish I had also the option to bridge the network to the Ethernet port so that I could also access the machine from SSH or/and have it download the relevant packages from the internet.
**To summarize:** We fake display and keyboard. It would be great if we could also fake Ethernet access using the same network the bliKVM is using wirelessly.

ThomasVon2021, 2024-05-15:
Simply , do you want to share the kvm wifi network to controlled PC through a network cable between the kvm and the controlled PC？

Altares, 2024-05-15:
> Simply , do you want to share the kvm wifi network to controlled PC through a network cable between the kvm and the controlled PC？

Yes, acting as a bridge to let the PC be on the same network.

ThomasVon2021, 2024-05-16:
This should be feasible, it is a software feature that I have used under Windows, and I think it should also be possible under Linux.