blikvm/blikvm #167: [Bug] Bios Screen is green

glnces, 2024-05-19:
I updated to v1.4 and all of the sudden when I boot into the bios the screen is green:

[image]

But when I boot into the OS, the screen goes back to normal:

[image]

ThomasVon2021, 2024-05-20:
Suspected to be incompatible with BIOS resolution, what specific hardware do you need? Can you try to changed to a different BIOS resolution

glnces, 2024-05-20:
> Suspected to be incompatible with BIOS resolution, what specific hardware do you need? Can you try to changed to a different BIOS resolution

I swapped resolutions already. I have it set to auto, which outputs 1080p. The other option was 1024x768 which was also green.

glnces, 2024-05-20:
I'm using a 5700G and plugged right into the motherboard. The bios looked fine when I first tested the BliKVM v4 out of the box before updating.

ThomasVon2021, 2024-05-20:
The collection format of TC358743 is RGB format, and the BIOS screen should display normally. However, if it is not in RGB output mode, the collection will not be normal after TC358743.

ThomasVon2021, 2024-05-20:
which blikvm hardware do you use? v1 v2 v3 use tc358743 video card, v4 use ms2131.

glnces, 2024-05-20:
> which blikvm hardware do you use? v1 v2 v3 use tc358743 video card, v4 use ms2131.

I'm using the v4.

ThomasVon2021, 2024-05-20:
I searched on Google for 5700G as an AMD graphics card, right？ AMD graphics card, you can go to the AMD console to change the display mode to RGB

ThomasVon2021, 2024-05-20:
This is a bit strange because upgrading to v4 normally does not affect the image color part. Can you power off and restart v4 to see? Is it a normal color for BIOS to connect to the monitor?

glnces, 2024-05-20:
By upgrading, I mean that I updated the software version of the v4 unit to 1.4. I've only ever had the v4 unit. The out of the box configuration for the v4 before I updated it to 1.4 showed the bios with normal colors.

The 5700G is an AMD CPU with onboard graphics, which is why I'm plugging into the motherboard.

ThomasVon2021, 2024-05-21:
I know, this phenomenon is quite strange now because color collection is not related to software upgrades, and I'm not sure why lower versions work properly. Some customers have encountered this phenomenon before, and the basic reason is that the output format of BIOS is not RGB. After modification, it will work normally.
I estimate that even if you return the old version of BIOS now, it will still be abnormal. Github's release has a historical version. Because the output video format of BIOS is determined by edid, which is written in flash and will not be upgraded.
You have completely powered off v4 and tested it, but the BIOS is still green, right.

glnces, 2024-05-22:
I did power it off and the green screen came back. I've been using the unit with tailscale and navigating to the tailnet name url, just randomly I decided to test going to the local IP address and the bios screen color looked normal. Then once I swapped back to the tailnet URL the colors remained good. Very weird.