blikvm/blikvm #171: [Feature Request] [BliKVM v4] Auto overclock for Armbian

mareksagan, 2024-06-10:
Hi, could you please add performance mode for BliKVM v4 so that the connection is more stable and it automatically overclocks and works faster for the better experience?

ThomasVon2021, 2024-06-13:
Will performance mode have any effect as the current CPU usage rate is not high?

mareksagan, 2024-06-13:
Well, the device seems to start faster and it seems more responsive after setting the CPU clock to maximum and turning on performance mode in Armbian

ThomasVon2021, 2024-06-21:
Can you provide a method so that I can test it.

mareksagan, 2024-06-28:
A method?

ThomasVon2021, 2024-08-25:
What I mean is how you set up and tested it, can you share it? I will also follow your method for setting up and testing.

NeVa0Theresia, 2024-09-22:
> What I mean is how you set up and tested it, can you share it? I will also follow your method for setting up and testing.
Change /etc/default/cpufrequtils ? Governor & Freq

blikvm, 2026-11-05:
Can you try to update to 3.0.x to test again? better latency.