blikvm/blikvm #175: [Bug] Image/Stream is draggable

rwardlow01, 2024-08-01:
Please add: draggable="false"
to the img tag like this:
[image]
this will prevent dragging the image element when trying to do a left click drag on the remote computer

ThomasVon2021, 2024-08-02:
I wiil try this

ThomasVon2021, 2024-08-14:
[image]
The new version has not yet been released and the software has added this parameter. Thank you for your suggestion.

ThomasVon2021, 2024-08-25:
https://wiki.blicube.com/blikvm/en/flashing_os/
You can flash to the lasted image to test.

ThomasVon2021, 2024-08-26:
Upgrade to the latest version, and if there are any issues, you can reopen it.

closed 2024-08-26 by ThomasVon2021